node-opcua/node-opcua #554: OpenSSL certificate not generated on Arch

sanderd17, 2018-11-11:
**I'm submitting a ...**  (check one with "x")

 - (X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
I just installed node-opcua from scratch, and got an OpenSSL error:

```[sander@ATLPOR13 solanum]$ npm install node-opcua -s
 Creating default g_config file  /home/sander/solanum/node_modules/node-opcua-client/certificates/config.js
Done 
OpenSSL version :  OpenSSL 1.1.1  11 Sep 2018
########################################### OPENSSL ERROR ###########################################
CWD = /home/sander/solanum/node_modules/node-opcua-client
Command failed: "openssl" genrsa  -out "/home/sander/solanum/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem" -rand "/home/sander/solanum/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/sander/solanum/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd into RNG
139994928587264:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:88:Filename=/home/sander/solanum/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd

########################################### OPENSSL ERROR ###########################################
Command failed: "openssl" genrsa  -out "/home/sander/solanum/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem" -rand "/home/sander/solanum/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/sander/solanum/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd into RNG
139994928587264:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:88:Filename=/home/sander/solanum/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd

Done 
```
The `private` directory is indeed empty (nothing to feed into the RNG I suppose).

**Expected behavior**
The certificate should be generated, so the sample-server.js can work out-of-the-box.

**Minimal reproduction of the problem with instructions**
`npm install node-opcua -s`

**Please tell us about your environment:**
  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (X)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: x64 desktop
  * OS version: Arch Linux (latest)
  * openssl package is installed (v 1.1.1)

* **node-opcua version:**  : npm latest: 0.5.0

* **Node:** 
   `node --version` = v11.1.0

erossignon, 2018-11-13:
checking with Latest version of ArchLinux on docker
  ```Linux version 4.9.93-linuxkit-aufs (root@856d34d1168e) (gcc version 6.4.0 (Alpine 6.4.0) )```

Version of openssl used in this distro is  1.1.1 ( 11 Sept 2018)

For some reason, openssl command do not create rnd files anymore

openssl 1.1.1 has a known bug described here https://github.com/openssl/openssl/commit/0f58220973a02248ca5c69db59e615378467b9c8 found on 13/09/2018

kruux, 2018-12-19:
The problem remains for me with openssl 1.1.1a.
I originally had this error with node-red but installing the latest node-opcua standalone with npm gives the same error.
Is there a way for me to fix this manually for now?

OS: Arch Linux
Node version: v11.5.0
Openssl version: 1.1.1a  20 Nov 2018
Node-opcua version: 0.5.5

```console
> node-opcua-client@0.5.5 postinstall /home/eric/projekt/opcua-test/node_modules/node-opcua-client
> node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_1024.pem

 Creating default g_config file  /home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/config.js
Done 
OpenSSL version :  OpenSSL 1.1.1a  20 Nov 2018
########################################### OPENSSL ERROR ###########################################
CWD = /home/eric/projekt/opcua-test/node_modules/node-opcua-client
Command failed: "openssl" genrsa  -out "/home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem" -rand "/home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd into RNG
140386230698496:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:98:Filename=/home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd

########################################### OPENSSL ERROR ###########################################
Command failed: "openssl" genrsa  -out "/home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem" -rand "/home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd into RNG
140386230698496:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:98:Filename=/home/eric/projekt/opcua-test/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd

Done 

> node-opcua-server@0.5.5 postinstall /home/eric/projekt/opcua-test/node_modules/node-opcua-server
> node test_helpers/create_certificates.js certificate -s -o certificates/server_selfsigned_cert_2048.pem

 Creating default g_config file  /home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/config.js
Done 
OpenSSL version :  OpenSSL 1.1.1a  20 Nov 2018
########################################### OPENSSL ERROR ###########################################
CWD = /home/eric/projekt/opcua-test/node_modules/node-opcua-server
Command failed: "openssl" genrsa  -out "/home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/PKI/own/private/private_key.pem" -rand "/home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/PKI/own/private/random.rnd into RNG
139734117696000:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:98:Filename=/home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/PKI/own/private/random.rnd

########################################### OPENSSL ERROR ###########################################
Command failed: "openssl" genrsa  -out "/home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/PKI/own/private/private_key.pem" -rand "/home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/PKI/own/private/random.rnd into RNG
139734117696000:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:98:Filename=/home/eric/projekt/opcua-test/node_modules/node-opcua-server/certificates/PKI/own/private/random.rnd

Done 

> node-opcua-server-discovery@0.5.5 postinstall /home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery
> node test_helpers/create_certificates.js certificate -s -o certificates/server_selfsigned_cert_2048.pem

 Creating default g_config file  /home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery/certificates/config.js
Done 
OpenSSL version :  OpenSSL 1.1.1a  20 Nov 2018
########################################### OPENSSL ERROR ###########################################
CWD = /home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery
Command failed: "openssl" genrsa  -out "/home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery/certificates/PKI/own/private/private_key.pem" -rand "/home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery/certificates/PKI/own/private/random.rnd into RNG
140155437445632:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:98:Filename=/home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery/certificates/PKI/own/private/random.rnd

########################################### OPENSSL ERROR ###########################################
[... cut ...] m" -rand "/home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery/certificates/PKI/own/private/random.rnd into RNG
140155437445632:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:98:Filename=/home/eric/projekt/opcua-test/node_modules/node-opcua-server-discovery/certificates/PKI/own/private/random.rnd

Done
```

erossignon, 2019-02-08:
Any volunteer to investigate this issue ?

turnerth, 2019-02-08:
We ended up solving it.  (Thanks Swapna Vad).  By using a different build container in docker.  In our Docker we changed from 

FROM alpine

to

FROM node:10-alpine

down lower in the Dockerfile we have 

#install nodejs into the base image
RUN apk update && apk add nodejs-current && apk add openssl && \
    rm -f /var/cache/apk/*

RUN apk add -U nodejs-npm  && \
    rm -f /var/cache/apk/*

DaCookie4u, 2019-02-28:
Did some research since I can't use the docker solution, not sure if this is the right direction though.

The certificates seem to be generated by `test_helpers/create_certificates.js` of the installing node (like node-opcua-client, node-opcua-server, and node-opcua-server-discovery).

This helper points to `node-opcua-pki/bin/crypto_create_CA` which is a different repository. So I am not sure if the problem actually in this repository.

If I am on the right path the problem should be somewhere here: https://github.com/node-opcua/node-opcua-pki/blob/master/lib/crypto_create_CA.ts

But I am unable to make any sense out of this. Unfortunately I have not idea about javascript.

turnerth, 2019-03-11:
Hi, our more recent builds started failing.  This is because this file

node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd

Does not exist.  I know that you mentioned using OpenSSL 1.1.1 or even 1.1.1a but that does not work either.

I put a random number in that file manually and then ran the openssl command manually and it starts to work.

However, when running using 

npm install

it fails.  (even though I put a random file in the folder).  After the npm install, my random.rnd file is gone!  So, I can only assume something is clearing out that folder.

My suggestion is for node-opcua to add a random file in there itself, since I don't seem to be able to.

Here is the error when running npm install
OpenSSL version :  OpenSSL 1.1.1a  20 Nov 2018
 ########################################### OPENSSL ERROR  ###############################
Command failed: "openssl" genrsa  -out "/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem" -rand "/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd" 2048
Can't load /node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd into RNG
140282196806504:error:2406F079:random number generator:RAND_load_file:Cannot open file:crypto/rand/randfile.c:98:Filename=/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd

 ########################################### OPENSSL ERROR  ###############################

kruux, 2019-03-11:
turnerth try downgrading openssl to 1.0.x. That is what fixed it for me. I didn't test it thoroughly, but I'm pretty sure you can upgrade openssl back up to 1.1.x after you have completed the npm install. The problem seem to be that the new versions expect the rand file to exist when the -rand command is given, and fails if it's not there. Where as the old versions created the file instead.

turnerth, 2019-03-12:
Hi.  I was able to solve it with this changed in the dockerfile, which downloads an earlier version of openssl.  

    RUN apk update && apk upgrade && apk add --repository http://dl-cdn.alpinelinux.org/alpine/v3.8/main openssl=1.0.2r-r0 && \
    rm -f /var/cache/apk/*

bompi88, 2019-03-19:
@erossignon Can we use the `-writerand` argument if there's no rand file and the `-rand` option if the rand file exists? Do you know if the `-writerand` argument has been around for a while?

smnscp, 2019-04-16:
This is is fixed by the [latest release](/node-opcua/node-opcua/releases/tag/v0.7.2), right? Thanks, @erossignon!

dirkzp, 2019-07-17:
Hi, I have the same problem now on Ubuntu Server 18. Downgrading to OpenSSL 1.0.X is not an option. Any suggestions?

CTGControls, 2019-10-28:
I am using the standard docker image see my docker compose file below. I am getting the error "Error: cannot locate certificate file /data/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem". When I look in the folder " /data/node_modules/node-opcua-client/certificates/" on only have the file config.js and the folder PKI. I know this is some how a know issue but I don't know how it fix the problem. Can I generate the certificate myself, if so could you give me the OpenSSL command.

  node_red:
    image: nodered/node-red:latest-12
    restart: always
    environment:
      - TZ=	America/Chicago
    volumes:
      - node_red_data:/data
    ports:
      - 1880:1880
      - 55388:55388
    stop_grace_period: 2m
    deploy:
      replicas: 1
      update_config:
        parallelism: 1
        delay: 1m
      restart_policy:
        condition: on-failure
      placement:
        constraints: [node.role == manager]

CTGControls, 2019-10-28:
So I have got the certificate get generating with OpenSSL using:

openssl req -newkey rsa:2048 -new -nodes -x509 -days 5475 -keyout /zfspoola/docker/volumes/backend_node_red_data/_data/node_modules/node-opcua-server/certificates/PKI/own/private/private_key.pem -out /zfspoola/docker/volumes/backend_node_red_data/_data/node_modules/node-opcua-server/certificates/server_selfsigned_cert_2048.pem

but now I get the error:

CTGControls, 2019-10-30:
I have found my problem, the newest docker builds did not include openSSL. I did a PR and soon openSSL will be included. See my PR at https://github.com/node-red/node-red-docker/pull/157

closed 2020-09-18 by erossignon