node-opcua/node-opcua #555: Unable to do npm install after node-opcua clone

samminen, 2018-11-15:
Issue - 1: I did clone the software from https://github.com/node-opcua/node-opcua  and when I do npm install, it failed with below error.

Error:

C:\Edge\Solution\nodeopcua\node-opcua-master\node-opcua-master>npm install
npm ERR! Error while executing:
npm ERR! C:\Program Files\Git\cmd\git.EXE ls-remote -h -t git://github.com/dtao/lazy.js.git
npm ERR!
npm ERR! fatal: unable to connect to github.com:
npm ERR! github.com[0: 192.30.253.113]: errno=No such file or directory
npm ERR! github.com[1: 192.30.253.112]: errno=No such file or directory
npm ERR!
npm ERR!
npm ERR! exited with error code: 128

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\H198358\AppData\Roaming\npm-cache\_logs\2018-11-15T14_43_04_582Z-debug.log

C:\Edge\Solution\nodeopcua\node-opcua-master\node-opcua-master>

Please help to resolve it; 

Issue 2: How to run sample_server.js  and sample_client.js from this node-opcua project ?  these 2 files present under node-opcua-master\node-opcua-master\documentation\ path in git;

sanderd17, 2018-11-15:
For number 1, apparently you can't connect to github. Please check your internet connection and firewalls.

For number 2, better install https://github.com/node-opcua/node-opcua-sample or https://github.com/node-opcua/node-opcua-sampleserver which will pull node-opcua as a dependency.

erossignon, 2018-11-16:
This is a generic nodejs and npm issue:
Please read related article about how to use npm and nodejs  behind a corporate proxy:
for instance https://jjasonclark.com/how-to-setup-node-behind-web-proxy/