node-opcua/node-opcua #558: Create address-space by xml , ns=2 ?

ericccccccc, 2018-11-19:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 
**Current behavior**

i use UAModeler to create my address-space model.
this is my server code
`var opcua = require("node-opcua");`
`var nodeset_filename1 ="./info_model/device.xml";`
`var server_options = {`
 `  port: 4334, // the port of the listening socket of the server`
 `  resourcePath: "UA/device",`
`   nodeset_filename: [
     opcua.standard_nodeset_file,
     nodeset_filename1,
   ]`
`};`
`~`

this is a part of my device.xml

` <UAObject NodeId="ns=1;i=5002" BrowseName="1:Device">
        <DisplayName>Device</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=58</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
        </References>
    </UAObject>
`

i set my object ns =1, but when i execute the server and use opcua-commander to connect the server.
i saw my object ns =2 ..

[image]

then i found that namespaceArray under the object/server/  have three value~
[image]

is that when i create the is server it will generate a namespace called "urn:NodeOPCUA-Server-default" by itself?
how to fix the problem ?

thx~

erossignon, 2018-11-24:
Hi @ericccccccc ,
This is normal, node-opcua may apply a translation to the namespace Index,
Namespace index inside the nodeset2.xml file are only valid inside the xml file itself. they correspond to the namespace described in full name in the namespace array at the top of the xml file.
There is already a namespace with index 1 in the server, it correspond to the namespace dedicated to store the server own object. ( addressSpace.getOwnNamespace().namespaceIndex === 1)

In fact you should not make any assumption about the namespace index that a imported nodeset file will have when  creating a server address Space.
you should use the ```AddressSpace#getNamespaceIndex("http://myorganisation/mynamespaceName")``` to retrieve the translated namespaceIndex once imported.

pelis, 2018-11-26:
I guess it should be addressSpace.getOwnNamespace().index === 1
When I am looking into server_engine.js it appears that a namespaceIndex other than 1 is not possible, is that correct?

server_engine.js:
// register namespace 1 (our namespace);
    const serverNamespace = engine.addressSpace.registerNamespace(engine.serverNamespaceUrn);
    assert(serverNamespace.index === 1);

closed 2019-02-08 by erossignon