node-opcua/node-opcua #564: Does it the example code of  Alarm and Conditions  ?

ericccccccc, 2018-11-27:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - (x) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

[image]

is that any A&C example to understand how it work??
i'm confused about what is Event or Condition in OPCUA , how it Alarm to client ??
And what's different with Pub/Sub? i think they are same ~

thx for anyone helps!!

erossignon, 2018-11-27:
@ericccccccc please refer to OPCUA Specitficiation part 3 4 5 and 9  to read about Events, Condition and then Alarms in general.
Unit tests are also a good place to find example of use.

ericccccccc, 2018-11-29:
i've found there are some Event example (simple_server.js 、simple_client.js) in 
node-opcua/packages/node-opcua-samples/bin/ ~
 when i try , i can't get any event notify , but only value of variable changing.
  
 
[image]

is that when i subscribe the event with the Object, when something change under the Object (like the value of variable is changing)  i can get the event notify on the monitor ??
i've watched the spec, but still confused about the event..

closed 2019-12-06 by stale[bot]