node-opcua/node-opcua #570: OPCUAClient certificate requires 'subjectAltName'

pelis, 2018-12-07:
- ( ) bug report => search github for a similar issue or PR before submitting
 - ( x ) feature request
 
I tried to connect a client - which actually works fine, however, I have received a certificate that has no 'subjectAltName' in which case the client crashes in the function below.
Would be nice if that could be checked so that
e.tbsCertificate.extensions.subjectAltName.uniformResourceIdentifier[0];
does not crash...
It's jsut a minor issue - but good to know...

OPCUAClient.prototype._getApplicationUri = function () {
    // get applicationURI from certificate
    const exploreCertificate = require("node-opcua-crypto").exploreCertificate;

    const certificate = this.getCertificate();
    let applicationUri;
    if (certificate) {
        const e = exploreCertificate(certificate);
        applicationUri = e.tbsCertificate.extensions.subjectAltName.uniformResourceIdentifier[0];
    } else {
        const hostname = require("node-opcua-hostname").get_fully_qualified_domain_name();
        applicationUri = makeApplicationUrn(hostname, this.applicationName);
    }
    return applicationUri;

};

num-lock, 2018-12-08:
OPC UA requires a subject alt name by specification. It's listed as mandatory. You have to fix your certificate.

erossignon, 2018-12-09:
Can you describe how the client certificate has been created ?

OPCUA Specification requires that Certificate contains subjectAltName 
( Release 1.04 page 10 OPC Unified Architecture, Part 4)

> All OPC UA Applications require an Application Instance Certificate which shall contain the
following information:
° The network name or address of the computer where the application runs;
OPC Unified Architecture, Part 4 87 Release 1.04
° The name of the organisation that administers or owns the application;
° The name of the application;
° The URI of the application instance;
° The name of the Certificate Authority that issued the Certificate;
° The issue and expiry date for the Certificate;
° The public key issued to the application by the Certificate Authority (CA);
° A digital signature created by the Certificate Authority (CA).

and part 6:
> 6.2 Certificates
6.2.1 General
OPC UA applications use Certificates to store the Public Keys needed for Asymmetric
Cryptography operations. All SecurityProtocols use X.509 v3 Certificates (see X.509 v3)
encoded using the DER format (see X690). Certificates used by OPC UA applications shall also
conform to RFC 3280 which defines a profile for X.509 v3 Certificates when they are used as
part of an Internet based application.
The ServerCertificate and ClientCertificate parameters used in the abstract
OpenSecureChannel service are instances of the Application Instance Certificate Data Type.
6.2.2 describes how to create an X.509 v3 Certificate that can be used as an Application
Instance Certificate.
6.2.2 Application Instance Certificate
An Application Instance Certificate is a ByteString containing the DER encoded form (see X690)
of an X.509 v3 Certificate. This Certificate is issued by certifying authority and identifies an
instance of an application running on a single host. The X.509 v3 fields contained in an
Application Instance Certificate are described in Table 36. The fields are defined completely in
RFC 3280.

...
>
[image]

jwende, 2018-12-09:
Good to remember yourself about the usage of „shall“ and „must“ ....
Thanks
J.

erossignon, 2018-12-09:
Please be more precise ... can you help non native english speaker ?  "Shall" means "soll" (german) or "doit" (french) and expresses a requirement to my understanding, not an option.
(https://ell.stackexchange.com/questions/171983/shall-vs-must-in-technical-documents )

jwende, 2018-12-09:
Hint for German speaking readers: 
https://medtech-ingenieur.de/shall-should-must-will-may/

So „shall“ translates to „hast to be there/implemented/provided“ .... similar to „Must“
J.

erossignon, 2018-12-10:
Thank you @pelis,  
Node-OPCUA will now display a warning message and should not crash anymore .
Please make sure that certificate have subjectAltName , you can use node-opcua-pki to generate such certifcites ,btw.

closed 2018-12-10 by erossignon