node-opcua/node-opcua #571: OPCUA Pubsub (Part 14)

jwende, 2018-12-07:
Hi there,

thank you for the existing code. 
I wonder if you have any plans / ideas for implementing OPCUA pub/sub based on Part 14 of the specification?

Regards
Joerg

erossignon, 2018-12-09:
Pub/Sub is implementation in node-opcua is currently being considered... I'll contact you about this.

jwende, 2018-12-09:
Glad to provide any help - even if I‘m not a hard core coder 🤓
Cheers 
J.

closed 2020-03-09 by stale[bot]

erossignon, 2022-08-13:
NodeOPCUA PubSub Module is available as a commercial module from Sterfive. 
An evaluation version has been published as a npm modules

https://www.npmjs.com/package/node-opcua-pubsub-server
https://www.npmjs.com/package/node-opcua-pubsub-client

```` 
npm install node-opcua-pubsub-server
````
See https://node-opcua.github.io/tutorial/2022/02/16/node-opcua-pubsub-episode1.html for more info

Full version, including source -code and npm modules are available for customers through [Sterfive](mailto:contact@sterfive.com) 's private repository

JoelKrec, 2023-02-27:
Hi everyone,
it seems that there is some kind of problem of importing the EventEmitter inside both pubsub evaluation packages which prevents transpilling user written code from typescript to javascript with strict settings. Is TypeScript not supported or is that a mistake that will be fixed?