node-opcua/node-opcua #572: Missing event for session reestablished

psorowka, 2018-12-09:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - (x ) feature request

**Current behavior**
When a connection breaks and is resumed, the client correctly emits a `connection_reestablished` event. However, this is only really true for the TCP Connection / Channel, but not for the session. If a session has timed out during the connection break, the automatic session resumption process runs *after* the `connection_reestablished` event. If the session object is used prior to this process being completed, this will lead to errors.

**Expected behavior**
It would be useful to 
a) either move the event emission to a later point in the code, after the session activation/repair has been completed (https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client/src/client_base.js#L551 e.g. to line 564)
b) introduce a second event `sessions_reestablished` that is emitted after session activation/repair has been completed

**Minimal reproduction of the problem with instructions**
- create a minimal client with a short `requestedSessionTimeout` 
- break the connection between client and server (so that the session times out)
- use the session immediately in the handler:

```
const client = new OPCUAClient({
  /* ... */
  requestedSessionTimeout: 500,
  /* ... */
 })
await client.connect(/* ... */)
const session = client.createSession(/* ... */)
client.on('connection_reestablished', () => {
  console.log('Client connection has been reestablished')
  session.read(/* ... */) // --> will raise BadSessionIDInvalid error!
})

/* break the connection, manually or automatically */ 
```

**What is the motivation / use case for changing the behavior?**
I randomly stumble about this problem in the wild because I have to deal with many session timeouts. We have implemented a number of automatisms when connection losses and reconnects occur and randomly run into problems because we have no indications on when the sessions can be used again.

**Please tell us about your environment:**

  * (x ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua

erossignon, 2019-03-19:
```connection_reestablished``` is raised when the secureChannel has been recreated at the secure channel level.
There could be multiple sessions por channel and the operation to re-active the sessions and repair their subscriptions requires several round trips with the server until every is back to normal. 

I would rather emit an new dedicated event on the Client and the session object itself to indicate that the session has been repaired and is back to normal.

```
// session_reestablished event is sent after a session has been successfully repaired following a reconnection  

client.on("session_reestablished",(session) => {
});
session.on("session_repaired",() => {
});
```

psorowka, 2019-03-19:
That sounds totally reasonable.

This would really fix a number of issues, because exactly these "several
round-trips" is what you need to wait for before operating on the session
again (e.g. for writing, reading...)

closed 2020-04-01 by stale[bot]