node-opcua/node-opcua #576: Install error with npm on RasPi 3

hankie77, 2018-12-19:
**I'm submitting a

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
after npm install I've got the following:

(node:14401) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN deprecated delayed@1.0.1: no longer maintained
npm WARN deprecated sprintf@0.1.5: The sprintf package is deprecated in favor of sprintf-js.
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.14.79-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/ILS/node-opcua
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm WARN deprecated coffee-script@1.12.7: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/ILS/node-opcua/npm-debug.log
npm ERR! not ok code 0

**Expected behavior**
npm install without any error

**Minimal reproduction of the problem with instructions**
install step followed as described here
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**
RaspBi 3b+, Raspbian OS, npm was installed with sudo apt get install npm afterwords. OS and packages on the last state

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (x)  my request is related to node-opcua acting as a OPCUA SERVER 

  * (x) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
  
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` = 8.11.1

erossignon, 2018-12-19:
it looks like an npm issue,
https://github.com/npm/npm/issues/20896

please update npm with
```
npm install -g npm
```
and retry

hankie77, 2018-12-20:
hi erossignon,

I've found it out that it was a right problem. If I install it with sudo, it works. Sometimes I have to install some packages afterwords (like chalk) but if I do this it works.

closed 2018-12-20 by hankie77

empierre, 2020-03-10:
I have this error installing as root:

(node:4801) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.19.66+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install" "-g" "pm2"
npm ERR! cwd /
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /npm-debug.log
npm ERR! not ok code 0
dpkg: erreur de traitement de l'archive /var/cache/apt/archives/mydomoathome_0.2.42_all.deb (--unpack) :
 le sous-processus nouveau script pre-installation a retourné une erreur de sortie d'état 1
Des erreurs ont été rencontrées pendant l'exécution :
 /var/cache/apt/archives/mydomoathome_0.2.42_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)