node-opcua/node-opcua #578: Is LocalDiscovery with Multicast Extension supported?

msalafia, 2018-12-30:
Is Local Discovery with Multicast Extension supported? At a glance it seems to be supported but i cannot find any documentation or examples or brief explaination on how to use it.

Can you please provide me some more information about this feature? Thank you in advance

erossignon, 2019-01-04:
Yes, LDS with ME extension is supported on OPCUAServer.
you can either create and start a LDS+ME server or make sure that your node-opcua Server register itself to the  LDS or any remote Discovery server. Server can also publish themselves using ME.

erossignon, 2019-03-09:
could you elaborate about your need ? do you want to instantiate an LDS server or get one of your node-opcua registered itself to the LDS server ?

msalafia, 2019-03-09:
No special need at the moment. I just want to know the features of node-opc-ua about LDS-ME and how to use it.

closed 2019-06-24 by erossignon