node-opcua/node-opcua #582: Cannot find nodeId

salomondeep, 2019-01-10:
Hello,
I have a question regarding the OPC-UA server and client tutorials.
I am running both and the client successfully connects to the server, but cannot retrieve the value of a node existing in the namespace of the server. It return a null value with a description that says that it cannot find the desired nodeId.
[image]

erossignon, 2019-01-11:
The object with the corresponding nodeId doesn't exists on the server apparently.
(each server exposes a different set of object with different nodeId)
Can you disclose the source of your index.js file ?

salomondeep, 2019-01-14:
The index.js file is the client opc-ua from the tutorial-> [https://github.com/node-opcua/node-opcua/blob/v0.4.5/documentation/sample_client.js](url)

I'm trying to read the other 2 variables besides the free_memory and I cannot. It just tells me that the node doesn't exist.

salomondeep, 2019-01-14:
> The index.js file is the client opc-ua from the tutorial-> [https://github.com/node-opcua/node-opcua/blob/v0.4.5/documentation/sample_client.js](url)
> 
> I'm trying to read the other 2 variables besides the free_memory and I cannot. It just tells me that the node doesn't exist.

Solved it. It was an issue w/ the names of the variables. 
One other thing, the server closes because of a keepalive that does not receive. How do I solve this in order for the server not to close and run 'forever'.
[image]

erossignon, 2019-02-08:
ok closing then

closed 2019-02-08 by erossignon