node-opcua/node-opcua #584: How to add custom variable type with variable of other custom variable type

breedj, 2019-01-14:
I've added a custom variable type and like to use it in another variable type. How do I do that. I get an error with the code below. 
I'm trying to create a variable type (InterfaceType) with a variable of array of type DescriptorType, which is also a custom type.

        var descriptorType = namespace.addVariableType({
          browseName: "DescriptorType"
        });
        
        var descriptorType_ID = namespace.addVariable({
            componentOf: descriptorType,
            browseName: "ID",
            description: "Parameter identification number",
            dataType: "Int64",
            modellingRule: "Mandatory",
            value: { dataType: opcua.DataType.Int64, value: 0 }
        });
        
        var interfaceType = namespace.addVariableType({
          browseName: "InterfaceType"
        });
        var interfaceType_Parameter = namespace.addVariable({
            componentOf: interfaceType,
            browseName: "Parameter",
            description: "Parameter data required for peer-to-peer coordination",
            dataType: descriptorType,
            valueRank: 1,           // Amount of dimensions.
            //arrayDimensions: [7],   // Undefined length.
            //value: { dataType: descriptorType, arrayType: opcua.VariantArrayType.Array, value: [] },  ??
            modellingRule: "Mandatory"
        });

PaBraIgm, 2019-01-14:
You define a variable type, not a data type with "DescriptorType". And you can not defined an array of variables, the value attribute of a variable can contain an array.

I think you will need to define a structured data type for "DescriptorType" and use that in the variable type.

breedj, 2019-01-15:
That's a problem. I have no clue how to create a structured type. That's what I tried at first.

closed 2020-01-22 by stale[bot]