node-opcua/node-opcua #585: Does the client can publish to server?

ericccccccc, 2019-01-16:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( x) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

hi~is that opcua client can  publish msg to server?
i only found that the example which client subscribe , server publish the notification to client.
thank you~

ericccccccc, 2019-01-16:
> **I'm submitting a ...** (check one with "x")
>
> * ( ) bug report => search github for a similar issue or PR before submitting
> * ( x) feature request
> * ( ) support request => Please do not submit [...]

erossignon, 2019-02-08:
In fact , in opcua, servers are intended to talk to the underlying device and publish the exposed variable value to the monitoring client. not the other way around.

closed 2020-02-15 by stale[bot]