node-opcua/node-opcua #586: How to get node datatype using node opcua client dynamically

samminen, 2019-01-17:
I have node opcua client, which is capable to write value to server node. But while constructing nodesToWrite array, i need find what is the node datatype i want to write new value. How to find that property dynamically ? 

In the below code, I need to find datatype dynamically instead of hard coded to Double:  
  dataType: opcua.DataType.Double,

 nodesToWrite.push({
                                    nodeId: reference.nodeId.toString(), 
                                        attributeId: opcua.AttributeIds.Value,
                                        indexRange: null,
                                        value: { 
                                            value: { 
                                                dataType: opcua.DataType.Double,
                                                value: nodeNewValue
                                            }
                                    }
                                });

erossignon, 2019-02-08:
Similarly than #588  you could interogate the AttributeIds.DataType of the node:

```
function getDataType(session,nodeId,callback) {
    session.read({ nodeId: nodeId, attributeId: AttributeIds.DataType},function(err, dataValue) {
        if (!err) {
            if (dataValue.statusCode === opcua.StatusCodes.Good) {
                assert(dataValue.statusCode === opcua.StatusCodes.Good);
                const dataTypeNodeId = dataValue.value.value; // 
 
                session.getBuiltInDataType(dataTypeNodeId, function (err, dataType) {
                           return callback(null,browseName);
                });
            }
        }
        callback(err, DataType.Null);
    });
}
```
note that dataValue.value.value is of type NodeId and may represent a data type which has been derived from the basic dataType. 
Basic dataTypes are Double, Float, String, ExtensionObject etc...
NodeOPCUA conveniently provides an utility method to help you walk up the hiearchy of complex dataType to find the corresponding basic DataType  : this is getBuiltInDataType

erossignon, 2020-01-08:
@samminen  i presumed it helped. Let close now

closed 2020-01-08 by erossignon