node-opcua/node-opcua #587: Extension object decoding error on  SessionDiagnostics  expected size was 494 actual size was  486

samminen, 2019-01-17:
In my OPCUA client, when I crawel for nodes, after successful crawel, I am getting below warning. any resoultion ?

=========================================
WARNING => Extension object decoding error on  SessionDiagnostics  expected size was 494 actual size was  486

var crawler = new NodeCrawler(the_session); 

        crawler.on("browsed",function(element){
            if (nodeFriendlyName != undefined) {

                nodeFriendlyName.find((friendlyName) => {
                    
                    if(friendlyName == element.browseName.name) {
                        nodesList.push(element.nodeId.toString());
                        if (element.references.length >0 ) {
                            element.references.forEach((reference) => {
                               
                                nodesList.push(reference.nodeId.toString());
                            });                        
                        }
                    }
                });
            }
        });

        var nodeId = "ObjectsFolder";
        console.log("READ: now crawling object folder ...please wait...");
        crawler.read(nodeId, function (err, obj) {
           
            callback(err);
        });

erossignon, 2019-02-08:
I suggest that you fill the template form and describe the server that you use to connect your client.
I suspect that your server implements an old version of OPCUA that have a shorter SessionDiagnostics structure.

erossignon, 2019-03-03:
@samminen  any new on this ? Please provide missing information ...

closed 2020-03-09 by stale[bot]