node-opcua/node-opcua #595: How can I write OPC UA nodes with datatype is array

SHIRO123412, 2019-02-19:
This is my OPC UA server 

namespace.addVariable({
            organizedBy: device,
            browseName: "Temperature",
            dataType: "String",
         
            value: new opcua.Variant({ dataType: opcua.DataType.String, value: "initValue" })
        });

 namespace.addVariable({
            organizedBy: device,
            browseName: "Temperature22",
            dataType: "String",
            nodeId: "s=qccc",
            value: {
                get: function () {
                    return new opcua.Variant(
                        {
                            dataType: opcua.DataType.String,
                            arrayType: opcua.VariantArrayType.Array,
                            value: ["initValue", "asd"]
                        });
                }
            }
        });
////////////////////////////////////////////////////////////////////////////////////////////
and  I use c# client to write this the code is as follows
      bool result1 = opcUaClient.WriteNode("ns=1;i=1001", "9");
      String v1 = opcUaClient.ReadNode<String>("ns=1;i=1001");

      object[] test_arr = new string[] 
     {
            "q1", "q2"
     };
     string[] str_arr = new string[]
     {
                 "ns=1;s=qccc",
                 "ns=1;s=qccc",
      };
      bool result2 = opcUaClient.WriteNodes(str_arr, test_arr);
      string[] v2 = opcUaClient.ReadNode<string[]>("ns=1;s=qccc");
////////////////////////////////////////////////////////////////////////////////////////////

my result1 and result2 is true but my v2's value no update  v1's value is update 
I want update v2's value thank you

zicheee, 2020-07-07:
Hi @SHIRO123412 
I think the answer for your question is this code:

```javascript
let testArray = ['initValue01', 'initValue02'];
serverNamespace.addVariable({
    componentOf: device,
    browseName: 'testArray',
    dataType: 'String',
    value: {
        get: function () {
            return new opcua.Variant({
                dataType: opcua.DataType.String,
                arrayType: opcua.VariantArrayType.Array,
                value: testArray
            });
        },
        set: function (variant) {
            testArray = variant.value;
            return opcua.StatusCodes.Good;
        }
    }
});
``` 

After setting new values in e.g. UAExpert it looks like this:
[image]

zicheee, 2020-07-07:
After answering to this question I also have an array question:
I'd like to create an array of objects but I don't know how to transform the code above to achieve this. I need to create a structure similar to this picture:

[image]

I have a dynamic array and this array contains the object "Box_Item". Each Box item has two variables that I want to write. Is that possible with node-opcua?

erossignon, 2020-07-23:
@zicheee  yes it it possible , you will have to specify the correct dataType on the variable that matches the structure you want to expose and specify the arrayDimension property as well.
you'll find a example in chapter **11.3 client writing an extension object** of the [node-opcua by example](https://leanpub.com/node-opcuabyexample) book.

closed 2020-10-11 by erossignon

jordy-u, 2022-02-02:
> If it's possible to send data as an array from an OPCUA Client, is it be possible to receive that array in Wincc Unified? Are there any ways to receive an array in Wincc Unified?

I never tried it, but the update-notes of TIA Portal say that you need to activate the setting "_OPC UA Server Array index range access_". This is described under the 'WinCC' chapter, not 'WinCC Unified'. Maybe this answer is not relevant in your scenario, but it seemed like an explanation for your problem.

Source:
https://support.industry.siemens.com/cs/document/109784441/updates-for-step-7-v17-s7-plcsim-v17-and-wincc-v17?dti=0&lc=en-SE