node-opcua/node-opcua #599: BadNodeIdUnknown on known working tags

anthrogan, 2019-02-26:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

When using subscribe some tags intermittently come back as terminated due to `BadNodeIdUnknown` the tags that come back as terminated are random and 99% of the time a restart of the app finds them fine
**Expected behavior**

Subscribed to known working tags
**Minimal reproduction of the problem with instructions**

  1. Have list of working tags in array
  2. forEach loop to create sub/callback for each tag
 ```
 items.forEach((tag) => {
    let monitoredItem = theSubscription.monitor(
      {
        nodeId: opcua.resolveNodeId(tag),
        attributeId: opcua.AttributeIds.Value
      }, 
      {
        samplingInterval: 1000,
        discardOldest: true,
        queueSize: 10          
      },
      TimestampsToReturn.Both,
    );
    monitoredItem.on("changed", function(dataValue) {
      console.log(dataValue.toString())

    });

    monitoredItem.on("terminated", function () {
      console.log(monitoredItem.statusCode)
    });
  });
```
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Docker Container

    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: WinCC Connectivity Pack
     - Version:_________
     - Manufacturer/Software vendor: Siemens
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` = 0.5.8

erossignon, 2019-03-03:
Please provide a complete example so it can be investigated ..   for example your code snippet does't show what your application is doing with theSubscription.
You can try to adapt the https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_client.js  code to reproduce the issue.

anthrogan, 2019-03-03:
Hi @erossignon you are right this message was coming from WinCC, someone had been messing with the FB that supplies the DB in S7! I forgot to close this ticket! My apologies.

closed 2019-03-03 by anthrogan