node-opcua/node-opcua #600: Fixing vulnerability issues

gabrieledeluca, 2019-02-26:
Hello @erossignon,
the following [vulnerability issues](https://snyk.io/test/github/node-opcua/node-opcua) could be fix? 

Thanks in advance

erossignon, 2019-02-26:
do not hesitate to submit a pull request with a fix

erossignon, 2019-03-03:
Btw, there may be vulnerabilities issues in packages used for development purposes (there are tolerated here as they are self contained on the developer computer and not accessible from the outer world) however installing node-opcua from npm should have no vulnerabilities and should produce 

```` 
$ npm i node-opcua
npm WARN deprecated delayed@1.0.1: no longer maintained
[...]
+ node-opcua@0.5.8
added 205 packages from 95 contributors and audited 1124990 packages in 45.236s
found 0 vulnerabilities
```

@gabrieledeluca  can you confirm this on your end ?

erossignon, 2019-07-21:
vulnerability issues are fixed on the flow when discovered on published node-opcua package.
some tolerance exists for external packages only used in development.

closed 2019-07-21 by erossignon