node-opcua/node-opcua #602: Don't work node-opcua >= 0.5.0  to nwjs > 0.33.4 (node 11.0.0)

mariodiciccio, 2019-03-01:
How to title of this post, why?

The project don't works for version 0.5.0, but is the nodejs version of the problem?

* Device: Desktop
* OS version: MacOS 10.13.6

mikakaraila, 2019-03-01:
We are using electron with TypeScript and works well.

erossignon, 2019-03-03:
@mariodiciccio It would be good to get a more detailed description of the issue and steps by steps to reproduce the behavior that you have encountered.

gsabboz, 2019-03-06:
I'm facing the same problem as @mariodiciccio. I tried it with several versions of node-opcua and electron. 

Electron seems to freeze when importing the node-opcua v0.5.8 
Electron seems to start up, but only partially when importing node-opcua v0.4.1 

With version v0.41, from the moment a `new opcua.OPCUAServer` instance is created it's the same result as in version 0.5.8. And electron completely freezes.

I package it as .dmg, .snap, .deb but the same result occurs on several machines 

Mac Os 10.14.2
Ubuntu 14.04
A virtual machine with Ubuntu 14.04

To reproduce this I forked the electron-react-boilerplate and added the node-opcua package. https://github.com/gsabboz/electron-react-boilerplate

--> yarn install
--> yarn start

The Electron application will not start up. 
When removing  
`import { startOpcuaServer } from './opcua/opcua-server';`
`startOpcuaServer();`
from ./app/index.js
The Electron application will start.
Any idea what is happening?

fbalazsy, 2019-03-06:
I've got the sorta same problem,
Macintosh:
1. Install https://github.com/SimulatedGREG/electron-vue
2. create file with:
import * as opcua from 'node-opcua'

var subscription
var client
var session

const endpointUrl = 'yourUrl'

export async function main () {
  try {
    client = new opcua.OPCUAClient({
      connectionStrategy: {
        maxRetry: 500,
        initialDelay: 200,
        maxDelay: 10 * 100
      }
    })

    client.on('backoff', () => console.log('retrying connection'))  //Here goes only logo, has to be event

    await client.connect(endpointUrl)

    session = await client.createSession()

    const browseDescription = {
      nodeId: 'RootFolder',
      browseDirection: opcua.BrowseDirection.Forward
   }

    const browseResult = await session.browse(browseDescription);
}  catch (err) {
    console.log('Error !!!', err)
  }
}
3. run in electron main process:
import * as opc from './pathToYourFile'
app.on('ready', () => {
  opc.main()
})

 Works fine, when you in dev, but when you build it, it will run many processes and won't start.
Linux:
Won't even start.
I tried it on NW.JS also.

mikakaraila, 2019-03-07:
Of course you should use async function in main not to block execution...

fbalazsy, 2019-03-07:
I am sorry but I don't get your point, is that a bad approach?

erossignon, 2019-03-09:
see also #607

erossignon, 2019-03-17:
Can someone confirm that this issue is now fixed with version 0.6.0 ? 
This will confirm also that this was a duplicate of #607

fbalazsy, 2019-03-17:
Working on darwin, linux should be fine then.

erossignon, 2019-03-17:
Thank you @fbalazsy , Let's close and label this issue as a duplicate of #607

closed 2019-03-17 by erossignon