node-opcua/node-opcua #606: Client doesn't reconnect.

mdameda, 2019-03-06:
**I'm submitting a ...** 

- (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
- Client make a connection successifully
- network cable were unplugged for 10sec
- network cable were plugged again
- Client didn't reconnect and every request produce this log:

` Timeout .... waiting for response for  ReadRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=2245089672
 timeStamp                     /* UtcTime                          */: 2019-03-06T08:30:10.003Z
 requestHandle                 /* IntegerId                        */: 37               0x25
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: 
 timeoutHint                   /* UInt32                           */: 0               0x0
 additionalHeader              /* ExtensionObject                  */: null
};
`
caused by Transaction timeout

**Expected behavior**
I would have expected a re-connection to be made.

**Minimal reproduction of the problem with instructions**
- Create new opc client and trigger a request every 2 seconds (for example)
- unplug network cable for 10sec
- plug network cable

**What is the motivation / use case for changing the behavior?**
It should be possible to reconnect to the opc ua server.
  
**Please tell us about your environment:**

  * (x ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Dell latitude
  * OS version: 
    
    - ( ) Windows : version : _________
    - (x ) Linux   : version : Ubuntu 18.04.1 LTS
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: PLC
     - Version: Siemens S7 1500
     - Manufacturer/Software vendor: Siemens
     - link : https://
     
    
* **node-opcua version:**  :
v0.5.8

* **Node:** 
   `node --version` = 10.15.0

erossignon, 2019-03-09:
Hi ,
If  you are using polling to read the state of a variable , you'll need to detect the disconnection by setting up an event handler on the "connection_lost" event so that you can suspend the transaction with the server.
you'll then can reanable the transaction by settingup a event handler on the "connection_restablished" event.

Alternatively, recommend that you use Subscription to monitor  modification of a server variable instead of polling read.
Nodeopcua has a build in mechanism to deal with disconnection and the server will be able to provide also the missing values that have changed during the  connection failure.

mdameda, 2019-03-11:
> by setting up an event handler on the "connection_lost" event so that you can suspend the transaction with the server.
you'll then can re-enable the transaction by setting up a event handler on the "connection_restablished" event.

Ok,  do you have an example how to ' re-enable the transaction' ?

I thought the correct event was timed_out_request: 
`this.client.on('timed_out_request', () => ...`  
but actually it did not help me.

> Alternatively, recommend that you use Subscription to monitor modification of a server variable instead of polling read.

I already have a ClientSubscription , together with the polling, but the reported problem also persists with the subscription. Here my subscription:
```   
let sub = new ClientSubscription(this._session, {
    requestedPublishingInterval: 500, 
    requestedLifetimeCount: 5, 
    requestedMaxKeepAliveCount: 1, 
    maxNotificationsPerPublish: 1, 
    publishingEnabled: true,
    priority: 10,
});
```
Together with the subscription I have a polling of 2 seconds to read a couple of Struct data, but after unplug/plug of net cable i still have the timeout without a reconnection (i have to kill my application...thing that i don't want to do in my production environment).
Thanks

erossignon, 2019-03-16:
What are the connectionStrategy used during the OPCUAClient creation ?
(connectionStrategyUsed also affects reconnection)

mdameda, 2019-03-17:
This is my connection strategy:

```
connectionStrategy: {
                maxRetry: 0,
                initialDelay: 4000,
                maxDelay: 10000,
                randomisationFactor: 0,
            }
```

Two questions:

1- maxRetry: 0 consists in an infinite reconnection attempt ?
2- On production environment what's the best configuration of **connectionStrategy** and **ClientSubscription** properties ?
For example, on subscription i used :
``
requestedLifetimeCount: 5, 
requestedMaxKeepAliveCount: 1
``

 but i saw a your old message where you sponsored this conf:

``
requestedLifetimeCount: 200, 
requestedMaxKeepAliveCount: 20
``

erossignon, 2019-03-17:
Hi Michele,

I did anticipate that the issue might come from the maxRetry value and started working on a fix.

Regarding question 1:

```maxRetry: 0``` means that there will be only one attempt to connect, and no retry; therefore initial connect will fail with an error if the server is not available.
It occurs that until version 0.6.0 maxRetry was also used for the reconnexion attempt, hence the behavior that you have seen.
I have today modified the reconnexion code to make sure that node-opcua now retries the connection indefinitely regardless of the maxRetry value of the connectionStrategy.

As a temporary workaround, I suggest that you set maxRetry to a very large value => 1000000 or so so that  it gives enough try for the reconnexion attempt.
Once again , in version 0.7.0, the reconnexion attempts will be infinite.

Also 0.7.0 will allow maxRetry to be set to -1 so that initial connection can infinitely try to connect.

ClientSubscription:

requestedPublishingInterval: 1000,
requestedLifetimeCount: 200, 
requestedMaxKeepAliveCount: 20

means : 
 - subscription will publish data notifications changes every 1000 ms ( 1 seconds)
 - requestedMaxKeepAliveCount = 20 , if no data modification is available the server will send a empty Publish Response every time 20 publishingIntervals have elapsed:
    20 * 1000 ms = every 20 seconds,
(Make sure that your session timeout is greater than this value)

- requestedLifetimeCount = should be a large value if you want to deal with long disconnexion period without loosing data.
- requestedLifetimeCount  controls the time the subscription will remain active and keep recording data if the connection between the client and server has been broken.
It give the opportunity for the client to reconnect to the same session and retrieve data that has been recorded on the server side but could not be sent to the client due to the connection break.
If the server is not able to serve the notification change to the client after requestedLifetimeCount * publishingInterval  , then the server is allowed to drop the subscription and unprocessed notification to save memory and ressources.
Obviously this could create a drain of resource on the server if your set requestedLifetimeCount  too high. The specification requires that requestedLifetimeCount  is at least 60 time bigger than requestedMaxKeepAliveCount 

Please refers to the OPCUA specification part 4 (5.13 Subscription Service Set) for more information.

mdameda, 2019-03-17:
Thanks for your prompt reply, I do some more tests and I will do it when 0.7.0 will be available.

psorowka, 2019-03-19:
@erossignon I think in all reconnect problems, you should not miss the issue described in #572 

Especially your comment 

> you'll then can re-enable the transaction by setting up a event handler on the "connection_restablished" event.

I think is not perfectly valid. `connection_restablished` is emitted before sessions are repaired. As the event emitter is working synchronously, any "transaction re-enabling" might thus be performed before the session is valid.

closed 2019-03-28 by erossignon