node-opcua/node-opcua #607: Crash when using node-opcua server inside electron on Mac and Linux

ptillemans, 2019-03-07:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

When importing *opcua-server* in an electron app.

```
import { startOpcuaServer } from './opcua/opcua-server';
```

It causes the program to never end importing dependencies and go into an endless loop of starting new electron applications till memory runs out.

**Expected behavior**

App continues with main routine after importing and display the UI. This is for the UI of an industrial control application which is running on equipment on the shopfloor.

**Minimal reproduction of the problem with instructions**

  1. create a minimal electron app. e.g. https://github.com/electron-react-boilerplate/electron-react-boilerplate
  2. add a dependency on opcua : yarn add node-opcua
  3. and an import : import { startOpcuaServer } from './opcua/opcua-server';
  4. package the app for linux : yarn package-linux
  5. install the package: sudo dpkg -i release/electron-react-boilerplate_0.17.1_amd64.deb     
  6. run the app : /opt/ElectronReact/electron-react-boilerplate 
  7. observe that the app is hanging and no output is generated.
  8. open another terminal and run *top* and observe the electron processes breed like rabbits.
  
What happens is that the *fqdn* module used by node-opcua-hostname package depends on an old version of shelljs. Shelljs uses *process.execPath* to find the name of the *node* executable to run small dynamically generated javascript scripts to run system commands. In most cases the *node* executably is the top level process and this works fine. For electron apps, this actually points to electron app which wraps node and starts a browser and the 2 javascript enviroments for the frontend and backend. 

When the fqdn library is used during imports it, it tries to call *hostname* by this method by executing the script with call to the program pointed to by *process.execPath* mistakensy assuming it points to the node interpreter. This keeps starting the electron app and initializing the app till memory runs out.

The fqdn library and shelljs libraries are only used from node-opcua-hostname package, and can be relatively easy be replaced with *os.hostname()* plus a dns lookup on Mac and Linux.

Windows is not affected, as this calling of *hostname* only happens on unix platforms as for windows it is derived differently.

**What is the motivation / use case for changing the behavior?**

Blocks deployment of OPC/UA to the shopfloor for our company.

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (x)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Xpeqt XTD Automatic Test Equipment
  * OS version: Ubuntu 14-04 and 18.04
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : Ubuntu 18.04
    - (x) MacOs   : version : Sierra
    - (?) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: see above, it is a server
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  : "0.5.8"

* **Node:** 
   `node --version` = v10.15.2  but probably not relevant

it happens with various electron versiong going back to 2.0.x

fbalazsy, 2019-03-07:
Same behaviour is explained in this issue:
https://github.com/node-opcua/node-opcua/issues/602

Could you please tell us versions of NodeJS, Electron or node-opcua in which you get this to work ?

fbalazsy, 2019-03-07:
Same here:
https://stackoverflow.com/questions/54516256/require-node-opcua-in-electron-freezes-the-electron-instance

mikakaraila, 2019-03-07:
Works with these versions at Windows 10:
"electron": "^4.0.2",
"node-opcua": "^2.0.0-alpha.10",

ptillemans, 2019-03-07:
It will always work on Windows as it has an alternative implementation to get the fully qualified hostname.

On unix  fqdn uses shelljs to execute 'hostname -f' to get the FQDN. Because fqdn is old, it depends on an old version of shelljs which assumes that process.execPath always points to the node executable which is not the case when running as an electron app.

The problem only manifests itself when using prebuilt apps which are installed as a .dmg file or .deb file. 

It does not occur when running in development.

fbalazsy, 2019-03-07:
Cannot wait for merging to test it !

ptillemans, 2019-03-09:
Thanks for the fix. We'll test when we get back to the office on monday.

ptillemans, 2019-03-11:
Tested with our test project and fix is effective. No more crash with 0.6.0. 

Testing with the main project will be done during this week but I am confident the issues is resolved.

ptillemans, 2019-03-11:
Gregory confirms it works with our target application locally, installed on ubuntu with a debian package, and installed on Mac with a .dmg package. Great work!

closed 2019-03-12 by erossignon