node-opcua/node-opcua #612: Is the logic of HasOrderedComponent ReferenceType implemented?

msalafia, 2019-04-08:
Is the semantic behind the HasOrderedComponent referenceType implemented by node-opcua?
The logic behind this ReferenceType should allow to return the same order of component after a Browse of the node. Is it implemented?

erossignon, 2019-04-09:
What the specs says:

OPC Unified Architecture, Part 3  Release 1.04 page 57

>7.9 HasOrderedComponent ReferenceType
The HasOrderedComponent ReferenceType is a concrete ReferenceType that can be used
directly. It is a subtype of the HasComponent ReferenceType.
The semantic of the HasOrderedComponent ReferenceType – besides the semantic of the
HasComponent ReferenceType – is that when browsing from a Node and following References
of this type or its subtype all References are returned in the Browse Service defined in Part 4
in a well-defined order. The order is Server-specific, but the Client can assume that the Server
always returns them in the same order.
There are no additional constraints defined for this ReferenceType

the key bit is  `all References are returned in the Browse Service defined in Part 4
in a well-defined order. The order is Server-specific `.

Nothing special has been added to support this semantics,  node-opcua will always return components in the same order if the same browse request is sent multiple time.
However the order may be altered if new components or references are added to the node at runtime. The reason for this is that references are stored in hashmap rather than array for performance reason.

So the answer is rather no.

msalafia, 2019-04-09:
@erossignon Thank you for the answer. I close the issue.

closed 2019-04-09 by msalafia