node-opcua/node-opcua #614: Client reconnect issue with Node 0.7.2 version

samminen, 2019-04-12:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
When I upgraded my Node-opcua package to version 0.7.2 and if I stop OPC server and wait for 5 min and my client is not notified with "backoff" or  "connection_reestablished" or "close" events.

Is there any re connection event notifications changed ? should I hook on to a different event emitter to get "reconnect attempting" notification ?

I am using SECURITY_MODE = NONE and SECURITY_POLICY = None when client connect to server.

**Expected behavior**
I am expecting opc client to receive "backoff" notification and it should do a retry attempt. 

**Minimal reproduction of the problem with instructions**

1. start OPC server and then start opc client
2. client will connect to OPC server
3. stop OPC server and client should be notified to "backoff" notification and should do a retry attempt.
  ...
  
**What is the motivation / use case for changing the behavior?**
This is very basic test to connect to to opc server

**Please tell us about your environment:**

I am running my test on WIN 10, Node OPCUA version 0.7.2

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =0.7.2

erossignon, 2019-04-25:
Can you provide more info about client and server , from which vendor ?  Did you write client or server or both ?

samminen, 2019-04-26:
Issue resolved; I created client instance and wired all client.on(events) and again after few steps i created new client instance. Since I wired my reconnect event for first client instance and I  re-initializing my client object, obviously on event callback will not be triggered. 

Now I managed to have one time client object initialization, everything works perfect. I fixed it yesterday.

Thank you for your immediate action on this question;

closed 2019-04-26 by erossignon