node-opcua/node-opcua #617: Cann't install node-opcua

VladBraz125, 2019-04-25:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - (x ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * (x ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: __PC___
  * OS version: _____
    
    - (x ) Windows : version : ___7 Pro______
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =

[image]

erossignon, 2019-04-25:
Seems to be with a corrupted package deasync@^0.1.7.
Can anyone else reproduce this as well ?

VladBraz125, 2019-04-26:
The Result of  npm i deasync@^0.1.14 is
npm ERR! code E403
npm ERR! 403 Forbidden: deasync@^0.1.14

VladBraz125, 2019-04-26:
[image]

erossignon, 2019-04-26:
What is the nodejs version ? may be it's too old.
Are you on Win32 or 64 ?

VladBraz125, 2019-04-26:
I work with win7 pro 64bit and have a newerste nodejs version. I am until 7.05 in holliday and have no possibility to check somethings.

erossignon, 2019-04-26:
Enjoy your holiday, you do not need to answer straightaway. When you'll be back, let me know the exact answer to your question.. expecially the exact version number of the "newest nodejs version".

erossignon, 2019-05-30:
@VladBraz125  can you please confirm that this issue is now fixed in 0.7.2 and in 2.0.0 as well ?

erossignon, 2019-06-24:
Issue seems to be stalled : closing

closed 2019-06-24 by erossignon

biancode, 2019-07-11:
deasync needs python-dev to be installed

erossignon, 2019-07-11:
deasync is not used anymore in the most recent versions of  node-opcua aka 0.7.2 and  2.0.0 onwards

biancode, 2019-07-11:
@erossignon hm .. we had that problem today with 0.7.4 and Nodejs 10.16.0

erossignon, 2019-07-11:
interesting ... please try ```npm ls > tree.txt``` and attach the file here