node-opcua/node-opcua #618: Rfid Scan Event Results Array is Null

ajcanterbury, 2019-04-26:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

Monitoring RFID Scan Result event from Simatic RF685 RFID Scanner. Event comes back as a datatype ExtensionObject with a Null array under value.

**Expected behavior**

Other tools (e.g. UA Expert) give the expected RfidScanResult array of Scan Data values. 

**Minimal reproduction of the problem with instructions**

  1. Subscribe to and monitor an AutoIDScanEventType -> RfidScanEventTpe (e.g. 3:ScanResult)?
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Siemens Simatic RF685
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - (x) MacOs   : version : 10.14
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  : 0.7.2

* **Node:** 
   `node --version` = 10.15

erossignon, 2019-10-06:
@ajcanterbury is it still the case with latest version ? ( I mean node-opcua >= 2.1.9 )

ajcanterbury, 2019-10-09:
I'll get back to you on this, I don't have the device currently to test on but should have it soon. I'll run some other extension object tests in the meantime.

closed 2020-10-16 by stale[bot]