node-opcua/node-opcua #621: node-opcua > 6.0 library takes ten times longer to load

ajcanterbury, 2019-04-30:
Updating from 5.8 to 6.0 (also affects the current 7) introduced a delay in the inclusion of the node-opcua library for me (node 10.15). Went from roughly 500ms to 5.5s. Not sure yet what is causing the issue.

Running `t=Date.now(); require('node-opcua'); Date.now()-t` produces:
> (node:50805) [DEP0097] DeprecationWarning: Using a domain property in MakeCallback is deprecated. Use the async_context variant of MakeCallback or the AsyncResource class
> 5428

erossignon, 2019-04-30:
Please fill the template.

erossignon, 2019-04-30:
On my side:
nodejs version: 
``` 
$ node --version
v10.15.3
````
os: 
```
$ node -e "console.log(process.arch, process.platform)"
x64 win32
```

with node-opcua@0.5.8:
```
node -e "t=Date.now(); require('.'); console.log(Date.now()-t)"
3295
```` 
with node-opcua@0.6.0:
```
$ node -e "t=Date.now(); require('node-opcua'); console.log(Date.now()-t)"
2686
```

with node-opcua@0.7.2
```
$ node -e "t=Date.now(); require('node-opcua'); console.log(Date.now()-t)"
2720
```

ajcanterbury, 2019-04-30:
And from me running your exact config except for the OS:
nodejs version:

```
$ node --version
v10.15.3
```
os:

$ node -e "console.log(process.arch, process.platform)"
`x64 darwin`
(OSX 10.14.4)

with node-opcua@0.5.8:

```
$ npm init
$ npm install node-opcua@0.5.8
$ node -e "t=Date.now(); require('node-opcua'); console.log(Date.now()-t)"
523
```
with node-opcua@0.6.0:

```
$ npm install node-opcua@0.6.0
$ node -e "t=Date.now(); require('node-opcua'); console.log(Date.now()-t)"
5470
```
with node-opcua@0.7.2

```
$ npm install node-opcua@0.7.2
$ node -e "t=Date.now(); require('node-opcua'); console.log(Date.now()-t)"
5430
```
with node-opcua@next

```
$ npm install node-opcua@next
$ npm dedupe
$ node -e "t=Date.now(); require('node-opcua'); console.log(Date.now()-t)"
434
```

erossignon, 2019-04-30:
The good news is that the  problem will resolved itself when node-opcua@next will be released.

ajcanterbury, 2019-04-30:
Going to test on a couple more platforms we are using and report back but any idea when next will be ready? No rush :)

erossignon, 2019-05-30:
version 2.0.0 is now out (https://github.com/node-opcua/node-opcua/releases/tag/v2.0.0) .

closed 2019-05-30 by erossignon