node-opcua/node-opcua #622: Can't read tag from server

torikachi24, 2019-05-01:
I have connected to my server from TiaPortal with s7-1500 is a OPCUA server..but i cant read data (value) from a nodeID although i can read status .this is my code and result
[image]
[image]

erossignon, 2019-05-01:
You need to prefix each path element with its namespace index, unless namespace index is 0;
```javascript
const ns1 =  1; // namespace of PCL_1
const ns2 =  3: // namespace of Memory
opcua.makeBrowsePath("RootFolder",`/Objects/${ns1}:PLC_1.${ns2}Memory`);
```

torikachi24, 2019-05-02:
thanks you...Now i can print browsePath of PLC_1 but i can't readVariableValue of tag "run"..how to do that.I'm just a begginer

erossignon, 2019-05-02:
```javascript
const ns1 =  1; // namespace of PCL_1
const ns2 =  3: // namespace of Memory
const browsePath = opcua.makeBrowsePath("RootFolder",`/Objects/${ns1}:PLC_1.${ns2}:Memory.${ns2}run`);
session.translateBrowsePath(browsePath,(err, result) {
   if (err) { return callback(err);
   const nodeId = result.targets[0].targetId;
   
   const nodeToRead = { nodeId, attributeId: opcua.AttributeIds.Value };
   session.read(nodeToRead, (err, dataValue) {
       if (err) { return callback(err);
       console.log(dataValue.toString());
  });
});

}),;
```

torikachi24, 2019-05-03:
i have try code like you. but dont know why it always fault 
"ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: Cannot read property '0' of null"
when i try to read tag run.how to fix it...and do you have any manual book about node-opcua. Thanks

closed 2019-05-04 by torikachi24

Ayush882, 2020-02-21:
Can You Please send Your Client Code as I am also not able to read the Namespace index and nodes.

erossignon, 2020-02-22:
@Ayush882 you will find a comprehensive example of how to do this in [my book "Node-OPCUA by Example"](https://leanpub.com/node-opcuabyexample)