node-opcua/node-opcua #625: Failed to create a tag while hosting an OPC UA Server.

rukechen, 2019-05-30:
**I'm submitting a ...**  (check one with "x")

 - ( x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

Create OPC UA Server with serveral node_id. Everything is good except node_id is something like ns=1;s=[test2]_Meta:O0.

Error: invalid namespace index ( out of bound)
    at assert (/opcua/node_modules/node-opcua-assert/dist/index.js:6:21)
    at AddressSpace.getNamespace (/opcua/node_modules/node-opcua-address-space/src/address_space.js:116:9)
    at AddressSpace.resolveNodeId (/opcua/node_modules/node-opcua-address-space/src/address_space.js:265:36)
    at AddressSpace.findNode (/opcua/node_modules/node-opcua-address-space/src/address_space.js:197:19)

It seems like the root cause is colon symbol because if I remove colon symbol I can create a OPC UA Server successfully.

**Expected behavior**

Should be able to create tag successfully.

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( X)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X ) I have installed node-opcua as a package ( using npm install )
  * ( X) I am using an application that uses node-opcua 
  
       - (X ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - (X ) Linux   : version : Ubuntu14.04
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_Ignition
     - Version: 8.0.2 rc1
     - Manufacturer/Software vendor: Ignition
     - link : https://inductiveautomation.com/
     
    
* **node-opcua version:**  : 0.7.0

* **Node:** 
   `node --version` = v8.15.1

erossignon, 2019-05-30:
I can't reproduce with this code. Am I missing something ?

```javascript
const opcua = require("node-opcua");

const server = new opcua.OPCUAServer({
    nodeset_filename: [ 
        opcua.nodesets.standard_nodeset_file
    ]
});

server.on("post_initialize",()=>{
    const addressSpace = server.engine.addressSpace;
    const mynamespace = addressSpace.getOwnNamespace();

    const obj = mynamespace.addObject({
        browseName: "MyObject",
        nodeId: "s=[test2]_Meta:O0.",
        organizedBy: addressSpace.rootFolder.objects
    });
    console.log(obj.toString());
    console.log(server.engine.addressSpace.getNamespaceArray().map(x=>x.namespaceUri).join("\n"));
});

server.start();

```

rukechen, 2019-05-31:
I think my description is not clear enough.
Here is my example.

`const opcua = require("node-opcua");
const NodeClass = require("node-opcua-data-model").NodeClass;

const server = new opcua.OPCUAServer(
    {
    port: 4840,
    nodeset_filename: [ 
        opcua.nodesets.standard_nodeset_file
    ]
});

server.on("post_initialize",()=>{
    tags = [{
                            browse_name: 'Devices',
                            parent: 'ns=0;i=85',
                            node_id: 'ns=1;s=5cef44f09c31dc004c8ae25a_sDevices',
                            origin_node_id: 'ns=1;s=Devices',
                            node_class: 'Object',
                            type_definition: 'ns=0;i=85' },
                          { 
                            browse_name: '[test2]',
                            parent: 'ns=1;s=5cef44f09c31dc004c8ae25a_sDevices',
                            node_id: 'ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]',
                            origin_node_id: 'ns=1;s=[test2]',
                            node_class: 'Object',
                            type_definition: 'ns=1;s=Devices' },
                          { 
                            browse_name: 'O0',
                            parent: 'ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]',
                            node_id: 'ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]_Meta:O0',
                            origin_node_id: 'ns=1;s=[test2]_Meta:O0',
                            node_class: 'Object',
                            type_definition: 'ns=1;s=[test2]' },
                          { 
                            browse_name: 'O0:0',
                            parent: 'ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]_Meta:O0',
                            node_id: 'ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]_Meta:O0/O0:0',
                            origin_node_id: 'ns=1;s=[test2]_Meta:O0/O0:0',
                            node_class: 'Variable',
                            data_type: 4,
                            interval: 100,
                            type_definition: 'ns=1;s=[test2]_Meta:O0',
                            permission: 3 }];
    createObjectsbyJson(tags);
});

createObjectsbyJson = function (tags) {
    let addressSpace = server.engine.addressSpace;
    let namespace = addressSpace.getOwnNamespace();
    tags.forEach(function (element) {
            if (element.node_class == NodeClass.Object.toString() && element.parent == "ns=0;i=85") {
                namespace.addFolder(
                    addressSpace.rootFolder.objects,
                    {
                        browseName: element.browse_name,
                        nodeId: element.node_id,
                        description: element.description
                    }
                );
            }
            else if (element.node_class == NodeClass.Object.toString()) {
                try {
                    namespace.addObject({
                        componentOf: addressSpace.findNode(element.parent),
                        browseName: element.browse_name,
                        nodeId: element.node_id,
                        typeDefinition: element.type_definition,
                        description: element.description
                    });
                } catch (err) {
                    console.log(err);
                }
            } else if (element.node_class == NodeClass.Variable.toString()) {
                try {
                namespace.addVariable({
                        componentOf: addressSpace.findNode(element.parent),
                        nodeId: element.node_id,
                        browseName: element.browse_name,
                        dataType: element.data_type,
                        accessLevel: element.permission,
                        description: element.description
                        })
                } catch (err) {
                
                }
            }
    });
}

server.start();`

[image]

rukechen, 2019-05-31:
By the way, I find that if I remove node "ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]_Meta:O0". Every thing is fine. I think maybe root cause is too many objects.
```
const opcua = require("node-opcua");
const NodeClass = require("node-opcua-data-model").NodeClass;

const server = new opcua.OPCUAServer(
    {
    port: 4840,
    nodeset_filename: [ 
        opcua.nodesets.standard_nodeset_file
    ]
});

server.on("post_initialize",()=>{
    const addressSpace = server.engine.addressSpace;
    const namespace = addressSpace.getOwnNamespace();
    namespace.addFolder(
        addressSpace.rootFolder.objects,
        {
            browseName: "Devices",
            nodeId: "ns=1;s=5cef44f09c31dc004c8ae25a_sDevices"
        }
    );
    namespace.addObject({
        browseName: "[test2]",
        nodeId: "ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]",
        componentOf: addressSpace.findNode("ns=1;s=5cef44f09c31dc004c8ae25a_sDevices")
    });
    /*namespace.addObject({
        browseName: "O0",
        nodeId: "ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]_Meta:O0",
        componentOf: addressSpace.findNode("ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]")
    });*/
    namespace.addVariable({
        
        nodeId: "ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]_Meta:O0/O0:0",
        browseName: "O0:0",
        dataType: "Boolean",
        accessLevel: 3,
        componentOf: addressSpace.findNode("ns=1;s=5cef44f09c31dc004c8ae25a_s[test2]"),
        })
    //console.log(obj.toString());
    console.log(server.engine.addressSpace.getNamespaceArray().map(x=>x.namespaceUri).join("\n"));
});

server.start();
```

rukechen, 2019-06-14:
Hi Erossignon:

Do you have any comment for this issue?

Thanks

closed 2020-06-22 by stale[bot]