node-opcua/node-opcua #630: Delay in publishing data acquired from modbus to opcua server

awadekaustubh, 2019-06-06:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
I am trying to publish data acquired from modbus TCP device to OPCUA server with following code.
`  val =  await mclient.readHoldingRegisters(2699, 18);`
      ` let l = 0;`
        `for(var j=0;j<val.data.length;){`
          `  var fulladdress = (2699 + j).toString();`
`            handler.ValueMap[rootname + fulladdress] = {`
`                v: new opcua.Variant({ dataType: "Float", value: val.buffer.readFloatBE(l) }),`
`                q: "good"`
 `           };`
`            l = l+4;`
`            j = j+2;`
 `       } `
` var handler = {
    ValueMap : {},
} `

The polling rate of modbus values is 200ms, and I am able to print them at the same rate.
With above code modbus values are published data to opcua server to their respective node id but, it takes more than 3 seconds to update the data.
I am using Prosys OPC UA Client to view the values.
Is there is any other way to publish values in opcua server which will have update time of 100-200ms? 

**Expected behavior**
The data published to opcua server should be available to opcua client in realtime at polling interval of 200ms

**Minimal reproduction of the problem with instructions**
I am using a modbus library and modbus TCP device to get the values and with above code to publish it to opcua server
  
**What is the motivation / use case for changing the behavior?**
Data Collected from any source like mqtt/modbus/s7 plc using node.js application linked to opcua server needs to be updated at realtime.

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( x)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Modbus TCP device
  * OS version: _____
    
    - ( x) Windows : version : 7
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: Prosys OPC UA Client
     - Version: 2
     - Manufacturer/Software vendor: Prosys
     - link : https://
     
    
* **node-opcua version:**  :
node-opcua version : 0.3.0

* **Node:** 
   `node --version` = 8.11.4

erossignon, 2019-06-09:
I don't understand the issue and how it relates to a 'bug' in node-opcua.
Please provide sample code of how you serve those variable through the OPCUA server , so problem can be 'reproducable`.

awadekaustubh, 2019-06-11:
[mbusHandler.txt](https://github.com/node-opcua/node-opcua/files/3275137/mbusHandler.txt)
I have attached the code for handling modbus transaction and putting modbus values to OPCUA Server using valuemap.

Can you go through it to check. May be it is not a bug but it is taking more time to reflect values in OPCUA server even if I am putting those value in interval of 200 miliseconds.

erossignon, 2019-06-11:
What are the subscription settings and monitoredItem settings on the client site ? ( publishInterval, samplingInterval etc )

awadekaustubh, 2019-07-12:
Sorry for delay in reply,
Publishing Interval is 500ms and sampling Interval is 50ms..
I have tried with lesser publishing interval, but it didn't work.

erossignon, 2019-08-19:
I would need to see the server code for review of the implementation on your side.

awadekaustubh, 2019-08-20:
I have attached the server code
[opcuaServer.txt](https://github.com/node-opcua/node-opcua/files/3518940/opcuaServer.txt)

erossignon, 2019-08-23:
It seems to be related to teh mbusHandler, support request are not handled in this forum..

closed 2019-08-23 by erossignon