node-opcua/node-opcua #632: Convertion inaccuracy when receiving float (32 bit) as a client

TBreidenbach, 2019-06-07:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
If a server writes float values (32 bit) and I read it with node opcua client, the value will be converted to double (64 bit) javascript variable with conversion deviation in decimals.
The server writes 13.37, the client reads 13.369999885559082.

**Expected behavior**
The server value and client value should be exactly the same. I checked it with UAExpert which shows the correct value.

**Minimal reproduction of the problem with instructions**
I wrote a standalone unit test that uses jest (V 24.1.0) and opcua ( V2.0.0)

Here is the code:

```javascript
/** @note This tests sets up a opc ua server, sets up a client and connects to the server */
/** @note The server will provide a float (not double!) variable which will be read */
/** @note by the client. Then the value set to server and read by client will be */
/** @note compared. Due to 32 bit float and 64 bit double (javascripts native datatype) */
/** @note the test will fail. */

import {
  OPCUAClient,
  OPCUAServer,
  DataType,
  VariantArrayType,
  Variant,
  AttributeIds,
  makeAccessLevelFlag
} from "node-opcua";

const server = new OPCUAServer({
  port: 4334, // the port of the listening socket of the server
  resourcePath: "/UnitTest", // this path will be added to the endpoint resource name
  allowAnonymous: true,
  buildInfo: {
    productName: "MySampleServer1",
    buildNumber: "7658",
    buildDate: new Date(2014, 5, 2)
  }
});

const client = OPCUAClient.create();

let testSession = undefined;

let exampleNodeId = undefined;

describe("OPCInterface", () => {
  const testValue = 13.37;
  const testValueName = "exampleFloat";
  let valueReceived = undefined;
  const valueExpected = testValue;
  let serverEndpointUrl = undefined;

  it("starts the server", done => {
    server.initialize(() => {
      const engineLocal = server.engine;

      const addressSpaceLocal = engineLocal.addressSpace;

      const namespace = addressSpaceLocal.registerNamespace(
        "float example bughunting"
      );
[... 686 lines cut ...]
| | +-- node-opcua-assert@2.0.0 deduped
| +-- node-opcua-variant@2.0.0
| | +-- node-opcua-assert@2.0.0 deduped
| | +-- node-opcua-basic-types@2.0.0 deduped
| | +-- node-opcua-data-model@2.0.0 deduped
| | +-- node-opcua-enum@2.0.0 deduped
| | +-- node-opcua-extension-object@2.0.0 deduped
| | +-- node-opcua-factory@2.0.0 deduped
| | +-- node-opcua-nodeid@2.0.0 deduped
| | +-- node-opcua-utils@2.0.0 deduped
| +-- node-opcua-vendor-diagnostic@2.0.0
| | +-- node-opcua-address-space@2.0.0 deduped
| | +-- node-opcua-assert@2.0.0 deduped
| | +-- node-opcua-constants@2.0.0 deduped
| | +-- node-opcua-server@2.0.0 deduped
| | +-- node-opcua-status-code@2.0.0 deduped
| | `-- node-opcua-variant@2.0.0 deduped

* **Node:** 
   `node --version` = v11.2.0

Am I doing something wrong or missing something here or is this a bug inside the decoding of the server message data stream?

Thank you in advance and best regards,
Thomas

erossignon, 2019-06-09:
In fact 13.37 in decimal is represented as 13.369999885559082 in float.
The fact that UAExpert shows 13.369999885559082 is because the float has been rounded for display purpose.

https://www.h-schmidt.net/FloatConverter/IEEE754.html

[image]

There is not much we can do about it.
I suggest that you either:
   - change the jest test with something like;
       ``expect(value).toBeCloseTo(number, numDigits)```
   - use Double instead of Float in your server

TBreidenbach, 2019-06-09:
Hey Etienne,

thanks for your reply.
Now I see that the root of the problem is that in float 32 the server will never be able to transmit values like 13.37 due to the inaccuracy of float 32 (and as you said double, too) in this range of values.
I thought the problem was receiving the value - but it already occurs when sending it.

When I change the test value from 13.37 to 13.36999988555908203125 the test passes - as you predicted.

So I will try to convince the PLC people to use either double or integers with offset.
Also I will insert some kind of rounding to my displayed values.

Please note that not my unit test is the problem but the requirement to display the same value in UA Expert and my application.

Thank you very much for your help.

Thomas

closed 2019-06-09 by TBreidenbach