node-opcua/node-opcua #634: Cannot create namespace with index

zicheee, 2019-06-11:
**I'm submitting a (x) bug report

**Current behavior**
With node-opcua (0.4.1) it was possible to create an object with any namespace index (in this example 3). I could then create nodes inside this namespace:
`
const device = addressSpace.addObject({
	organizedBy: addressSpace.rootFolder.objects,
    browseName: 'MY_OBJECT',
	nodeId: new opcua.NodeId(opcua.NodeIdType.STRING, '"MY_OBJECT"', 3)
});

addressSpace.addVariable({
	dataType: 'Int16',
	componentOf: device,
	browseName: 'MY_NODE_01',
	nodeId: 'ns=3;s="DB100_OPC_UA"."MY_NODE_01"',
	value: {
		get: function () {
			return new opcua.Variant({dataType: opcua.DataType.Int16, value: 10 });
		},
	}
});
`

When using UaExpert the object MY_OBJECT has the NamespaceIndex 3. The identifier is MY_OBJECT, the BrowseName is 0, "MY_OBJECT". 

Switching to the node-opcua (0.7.2) creating the Namespace with the code above is no longer possible. I tried the solution from #440 using code from the new documentation:
`
const device = namespace.addObject({
	organizedBy: addressSpace.rootFolder.objects,
	browseName: new opcua.QualifiedName({ name: "MY_OBJECT", namespaceIndex: 3}),
});

namespace.addVariable({
	dataType: 'Int16',
	componentOf: device,
    browseName: '3:MY_NODE_01',
	nodeId: 's="MY_OBJECT"."MY_NODE_01"',
    value: {
		get: function () {
			return new opcua.Variant({dataType: opcua.DataType.Int16, value: 10 });
		}
	}
});
`
But the result is different. With the code above the server won't start. The error is, that the namespace index is out of bounds. When removing the 3: before the node name the server starts, but the nodes are not inside the namespace index 3.

**Please tell us about your environment:**
I am using Ubuntu 18.04 with node v8.10.0
My Request is related to node-opcua acting as a OPCUA SERVER 
I have installed node-opcua as a package ( using npm install )

erossignon, 2019-06-11:
Namespace must be created before you can create object that relates to it. Please create a namespace .The bug and limitation was in version prior 0.4.4.
See https://github.com/node-opcua/node-opcua/releases/tag/v0.4.5 for description of breaking changes.

zicheee, 2019-06-12:
Hi @erossignon,

thank you for your response. I understand now that it was a bug before.

I cannot find any documentation on how to create a namespace. I know this question is more for stackoverflow but I think it is helpful here for others as well. If you can send me a link / code sample I can also create a documentation example.

Best regards
Michael

erossignon, 2019-06-12:
Usually, Server creates their own object instances in a dedicated namespace that node-opcua creates for you. You can access it
this way:
```
const namespace = addressSpace.getOwnNamespace();
```

zicheee, 2019-06-12:
Hi @erossignon 

I know this example but with this code I still cannot create a namespace with another index (e.g. 3). That is the important thing for me. I need this becuase I am using node-opcua for automated testing and the real OPC-UA server (Siemens S7) has their nodes in a namespace with index 3. With node-opcua v0.4.1 I could create the structure 100% identical to the Siemens S7 but with the current version I can't because I cannot find any example / documentation on how to create a new namespace with the index I need.

teropes, 2019-06-13:
I have exactly same use case. I think we should be able to control the application namespace index since that is not always 1 in real life scenarios, for example Siemens devices expose index 3.

Being able to mock real device (exact nodes, structures etc.) with node-opcua dev server is super useful for development / qa.

pelis, 2019-06-13:
Have you tried this?
var namespace2 = server.engine.addressSpace.registerNamespace("urn:Something:namespace2");
var namespace3 = server.engine.addressSpace.registerNamespace("urn:Something:namespace3");

When you call server.engine.addressSpace.registerNamespace() a new name space is appended. The index will be set to the namespace array index automatically.

In my case I have just added one namespace (namespace2) - this works for me.

zicheee, 2019-06-13:
@pelis 

Thank you very much for this comment. Those are the lines I was looking for.
I can simulate my S7 again. The Identifier and the IdentifierType are still different from the root (DB100_OPC_UA) object (don't know how to set / change) but the variables are now in the currect NamespaceIndex. That's most important for me.

Here is an example code (edited [sample_server.js](https://github.com/node-opcua/node-opcua/blob/v0.7.2/documentation/sample_server.js) from documentation):

```let opcua = require('node-opcua');

// Let's create an instance of OPCUAServer
var server = new opcua.OPCUAServer({
    port: 4850,
    buildInfo : {
        productName: 'MySampleServer1',
        buildNumber: '10',
        buildDate: new Date(2019,6,3)
    }
});

function post_initialize() {
	
  function construct_my_address_space(server) {
	  
    const addressSpace = server.engine.addressSpace;
    const unusedNameSpace = addressSpace.registerNamespace("unusedNameSpace");
    const spaceWithIndex03 = addressSpace.registerNamespace("DB100_OPC_UA");
        
    // declare a new object
    let device = spaceWithIndex03.addObject({
      organizedBy: addressSpace.rootFolder.objects,
      browseName: 'DB100_OPC_UA'
    });
    
    // add some variables 
    // add a variable named MyVariable1 to the newly created folder "DB100_OPC_UA"
    let variable1 = 1;
        
    // emulate variable1 changing every 500 ms
    setInterval(function() {
      variable1 += 1;
    }, 500);
	
    spaceWithIndex03.addVariable({
      componentOf: device,
      browseName: "MyVariable1",
      dataType: "Double",
      value: {
        get: function () {
          return new opcua.Variant({dataType: opcua.DataType.Double, value: variable1 });
        }
      }
    });
  }
  
  construct_my_address_space(server);
  server.start(function() {
    console.log('Server is now listening ... ( press CTRL+C to stop)');
    console.log('	port: 		' + server.endpoints[0].port);
    var endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
    console.log('	endpoint: 	' + endpointUrl );
  });
}

server.initialize(post_initialize);
```

closed 2019-06-13 by zicheee