node-opcua/node-opcua #635: Subscribed String Variable Quality not update correctly if timestamped_get is used

pbranchi, 2019-06-20:
**I'm submitting a ...**  (check one with "x")

 - ( X ) bug report
 - ( ) feature request
 - ( ) support request 

**Current behavior**
"String" Variable always seen in Good Quality when timestamped_get is used, returning value, statusCode and Timestamps. Int/Float variables displays correct quality Info

**Expected behavior**
"String" variables should behave the same as Int/Float

**Minimal reproduction of the problem with instructions**

  1. create a NODE OPCUA Server with following tags:
`const device = namespace.addObject({
            organizedBy: addressSpace.rootFolder.objects,
            browseName: "MyDevice"
        });
        
        // add some variables 
        // add a variable named MyVariable1 to the newly created folder "MyDevice"
        let variable1 = 'ciao';
        let quality1 = opcua.StatusCodes.Good;
        let ts1 = new Date();

        let variable2 = 1;
        let quality2 = opcua.StatusCodes.Good;
        let ts2 = new Date();
        
        // emulate variable1 changing every 500 ms
        setInterval(function(){  
            if (variable1 == 'ciao'){
                variable1 = 'addio';
                quality1 = opcua.StatusCodes.Bad;
            } else {
                variable1 = 'ciao';
                quality1 = opcua.StatusCodes.Good;
            }
            ts1 = new Date();
        }, 2000);

        setInterval(function(){  
            if (variable2 == 1){
                variable2 = 2;
                quality2 = opcua.StatusCodes.Bad;
            } else {
                variable2 = 1;
                quality2 = opcua.StatusCodes.Good;
            }
            ts2 = new Date();
        }, 2000);
        
        namespace.addVariable({
            componentOf: device,
            browseName: "MyVariable1",
            dataType: "String",
            value: {
                timestamped_get: function () {
                    let myDataValue = new DataValue({
                                value: new opcua.Variant({dataType: opcua.DataType.String, value: variable1}),
                                statusCode: quality1,
                                sourceTimestamp : ts1,
                                sourcePicoseconds: 0,
                                serverTimestamp: ts1,
                                serverPicoseconds: 0,
                            })
                    return myDataValue;
                }
            }
        });

        namespace.addVariable({
            componentOf: device,
            browseName: "MyVariable2",
            dataType: "UInt32",
            value: {
                timestamped_get: function () {
                    let myDataValue = new DataValue({
                                value: new opcua.Variant({dataType: opcua.DataType.UInt32, value: variable2}),
                                statusCode: quality2,
                                sourceTimestamp : ts2,
                                sourcePicoseconds: 0,
                                serverTimestamp: ts2,
                                serverPicoseconds: 0,
                            })
                    return myDataValue;
                }
            }
        });
`
  2. Start the server
  3. Subscribe to the tag above with a client such as Ignition, Softing Data Feed or FactoryTalk Live Data. This happens:
 
[image]
[image]

When Variable1 value is 'addio', quality should be Bad, not Good.

  
**What is the motivation / use case for changing the behavior?**
In order to communicate with some OPCUA Client, a Variant AND a Quality MUST always be returned. Using timestamped_get as explained above works fine, but "String" behaviour must be compliant to other types behaviour.

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (X )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device:  PC
  * OS version: _____
    
    - (X ) Windows : version : 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: Ignition
     - Version: 8.0
     - Manufacturer/Software vendor: Inductive Automation
     - link : https://inductiveautomation.com/ignition/whatsnew

     - Name: Softing Data Feed
     - Version:_________
     - Manufacturer/Software vendor: Softing
     - link : https://data-intelligence.softing.com/products/opc-software-platform/datafeed-opc-suite/
     
    
* **node-opcua version:**  :

2.0.0 but same problem tested with 0.7.2 and 0.7.3

* **Node:** 
   `node --version` = 10.15.3

erossignon, 2019-06-22:
thank you, I can reproduce it.
 this bug not only affects string variable but any array-like variable where numeric range could be used.

erossignon, 2019-06-29:
this should now be fixed in node-opcua@0.7.4  and will be also in next incremental build of node-opcua@2.0.x

closed 2020-06-29 by erossignon