node-opcua/node-opcua #636: Install error on Raspberrian

LaughtonAndrew, 2019-06-22:
I am submitting a bug report.
The install procedure at https://github.com/node-opcua/node-opcua/wiki/installing-node-opcua-server-on-raspberry does not work.

Everything up to $npm install is OK, but the install script gives the following output;
======================================================================
pi@raspberrypi:~/node-opcua $ npm install
(node:20293) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN deprecated delayed@1.0.1: no longer maintained
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.19.42-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/node-opcua
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm WARN deprecated typescript-eslint-parser@22.0.0: Deprecated: Use @typescript-eslint/parser instead
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/node-opcua/npm-debug.log
npm ERR! not ok code 0
pi@raspberrypi:~/node-opcua $ 

=======================================================================

STEPS TO REPRODUCE
Get latest version of full features Raspbian.   2019-04-08-raspbian-stretch-full
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install apache2 sqlite3 python php-db php-gettext wwwconfig-common mc openssh-server npm
    • Enable the ssh service by typing sudo systemctl enable ssh
    • Start the ssh service by typing sudo systemctl start ssh
Follow directions at https://github.com/node-opcua/node-opcua/wiki/installing-node-opcua-server-on-raspberry.

pi@raspberrypi:~/node-opcua $ node --version
v8.11.1
pi@raspberrypi:~/node-opcua $ npm -version
1.4.21
(node:20409) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
pi@raspberrypi:~/node-opcua $ git version
git version 2.11.0

erossignon, 2019-06-24:
I think the instruction to install latest version of nodejs are now out-dated in the wiki page.
Please look at https://github.com/blobsmith/raspberryTestNode/wiki/Node.js-installation-with-nvm-on-Raspberry-pi for instance to see if this steps works better..
Do not forget to update the wiki page and close this issue once you found and document the fix.

erossignon, 2019-06-26:
thkis may help as well to install LTS nodejs version from nodejs.org https://www.instructables.com/id/Install-Nodejs-and-Npm-on-Raspberry-Pi/

erossignon, 2019-07-11:
@LaughtonAndrew  any feedback ?

erossignon, 2019-11-14:
closing then

closed 2019-11-14 by erossignon