node-opcua/node-opcua #637: opcua client is connecting to server without verifying server certificate

jagadishpb89, 2019-07-04:
First, lot of thanks to you for writing node-opcua. Outstanding work!

**Current behavior**
When I connect opcua client to any opcua server without knowing or trusting server certificate, I am able to connect.

**Expected behavior**
Client without trusting the server certificate should not be able to connect to server.

**Minimal reproduction of the problem with instructions**
    var options = {
        endpoint_must_exist: false,
        securityMode: securityMode,
        securityPolicy: securityPolicy,
        defaultSecureTokenLifetime: 40000,
        certificateFile: certificateFile,
        privateKeyFile: privateKeyFile,

        keepSessionAlive: true,
        connectionStrategy: {
            maxRetry: 0
        }
    };

    const client = opcua.OPCUAClient.create(options);
Here I am not mentioning anything about trusted certificates at the client side. I am using self signed certificates. 
  
**What is the motivation / use case for changing the behavior?**
So should there be any option something like list which saves trusted server list  ? Is it also implemented ?

**Please tell us about your environment:**
  * () OS: debian
  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( ) I have installed node-opcua as a package ( using npm install )
        
  * Device: amd64
  * OS version: debian:stretch
   
   * Description of the other OPCUA system I am trying to connect to:
     - Name: node opcua server

    
* **node-opcua version:**  :
node-opcua@2.0.0

* **Node:** 
   `node --version` = v8.16.0

erossignon, 2019-07-08:
Correct, this verification is currently lacking.

limengsong, 2019-07-31:
@jagadishpb89 
Hi I am trying to connect an opc ua server. Requires a username and password, as well as a certificate. I don't know how the client should be configured. Please show your demo. Thank you.

limengsong, 2019-07-31:
@jagadishpb89 
```
        const client = new opcua.OPCUAClient({
            connectionStrategy: {
                maxRetry: 2,
                initialDelay: 2000,
                maxDelay: 10 * 1000
            },
            certificateFile : "./certificates/certs/uaservercpp.der",
            privateKeyFile: "./certificates/private/uaclientcpp.pem"
        });
```
How to build this client？

erossignon, 2020-06-13:
node-opcua@2.5.4 now performs certificate checks and passes the CTT certificats validation tests on server side.  
Still some work to be done to verify server certificate on client side.

erossignon, 2021-01-17:
node-opcua@2.29 : Client now provides comprehensive certificate verification (CA/ CRL , trusted, rejected etc ... )

closed 2021-01-17 by erossignon