node-opcua/node-opcua #638: Client side of official OPC UA Websocket+JSON Specification

Pro, 2019-07-08:
In the open62541 project we got a new PR which will add the official Websocket Transport specification aside of the default TCP. (See OPC UA Specification 1.04, Part 6, 7.5 Websockets)
https://github.com/open62541/open62541/pull/2872
In addition we already have the JSON encoding mostly implemented.

I guess the best chance to get an OPC UA client which implements the websocket transport + JSON Encoding is your project?!

Unfortunately I did not yet look into your implementation, but I'm asking if you already have some OPC UA client implementation supporting the official websocket transport and especially the official OPC UA JSON protocol mapping ?

Or can you estimate where and how long the extension of your code would take to add this feature? Maybe we can find some volunteer to implement it.

erossignon, 2019-07-08:
Hi Stefan,
JSON encoding à la OPCUA is partially implemented although not testing in real application.
websocket transport is not yet implemented and sterfive is seeking some funding sponsors to initiate the work (probably the longer part of the overall project)
It would be good if we could join effort on this.

Pro, 2020-07-08:
ping to prevent closing

closed 2021-07-21 by stale[bot]