node-opcua/node-opcua #640: Not able to reconnect once power to the opcua server is turned on

manjushreenimbalkar, 2019-07-10:
I'm submitting a ...**  (check one with "x")

 - ( x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
Not able to reconnect to the opcua server if the power to the server is disconnected

**Expected behavior**
Once the opcua server gets the power it should reconnect and start reading the values that were subscribed during the initial connection setup. 

**Minimal reproduction of the problem with instructions**

  1. Using WAGO opcua server  
  2. Created nodejs applicaiton using npm install  to connect to the opcua server
  3. Turn the power off to the opcua server.
  4. Turn the power on to the server back.
  5. Application gives warning : ClientSessionKeepAliveManager#ping_server  Connection Break
  ... start_reconnection index.js:52
 ... connection is reestablished after a connection break index.js:58
 ... reconnection process has been completed:  null 
  6. Nothing happens after this. It dosen't show me the values of monitored items.

My code
/*global require,console,setTimeout */
var opcua = require("node-opcua");
var async = require("async");
var Common = require('./Common');
var _config = require('./config/addressmap.json');
var deviceSetting;
var addressSettingList;
var header= {};

const INFO = "INFO";
const DEBUG = "DEBUG";
const ERROR = "ERROR";

var client = new opcua.OPCUAClient({
    keepSessionAlive: true,
    endpoint_must_exist: false,
    connectionStrategy: {
        maxRetry: 100000, 
        initialDelay: 2000,
        maxDelay: 10 * 1000,
    },  
});

deviceSetting = _config["DeviceSettings"];
addressSettingList = _config["AddressSettings"];
header.protocol = "opcua";
header.version = "1.0";
header.os = "tbd";
header.devicename = deviceSetting["deviceName"];
header.datetime =Date.now();

var endpointUrl =  deviceSetting["endpointUrl"]; 

var the_session, the_subscription;

async.series([

    // step 1 : connect to
    function(callback)  {
        Common.LogMessage(deviceSetting, DEBUG ,"connecting to endpoint: " + endpointUrl);  
        client.connect(endpointUrl,function (err) {
            if(err) {
                Common.LogMessage(deviceSetting, ERROR," cannot connect to endpoint :" + endpointUrl);  
[... 148 lines cut ...]
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - (x ) Windows : version : 10 pro
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: WAGO 787-1002
     - Version:_________
     - Manufacturer/Software vendor:WAGO
     - link : https://
     
    
* **node-opcua version:**  :
latest version 

* **Node:** 
   `node --version` = v10.16.0

erossignon, 2019-07-11:
It looks like the session is closed too early in your code...
You should let the client run without closing the session, nor disconnecting

manjushreenimbalkar, 2019-07-15:
I have removed the call for session close and disconnect still facing the same issue.

erossignon, 2019-07-19:
You would need to make sure that monitored item are created  in series . waiting for the previous one to be initialized before creating the next one. In your code, the server is bombarded with monitored item creation requests that may overflow the server capacity.

I would suggest that you use MonitoredItemGroup instead.
Monitored Item group allows you to create mutilple monitored item in a single transaction with the server:

```javascript
/* jshint esversion: 6 */
"use strict";
const opcua = require("node-opcua-client");

const resolveNodeId = opcua.resolveNodeId;

const endpointUri = "opc.tcp://localhost:48010";

(async () => {

    try {

        const options = {
            applicationName: "ClientBrowseNextDemo",
            endpoint_must_exist: false,
        };

        const client = opcua.OPCUAClient.create(options);

        client.on("backoff", (retry, delay) => {
            console.log("Backoff ", retry, " next attempt in ", delay, "ms");
        });

        client.on("connection_lost", () => {
            console.log("Connection lost");
        });

        client.on("connection_reestablished", () => {
            console.log("Connection re-established");
        });

        client.on("connection_failed", () => {
            console.log("Connection failed");
        });
        client.on("start_reconnection", () => {
            console.log("Starting reconnection");
        });

        client.on("after_reconnection", (err) => {
            console.log("After Reconnection event =>", err);
        });

        await client.connect(endpointUri);

        const session = await client.createSession();

        const subscription = await session.createSubscription2({
            maxNotificationsPerPublish: 1000,
            publishingEnabled: true,
            requestedLifetimeCount: 100,
            requestedMaxKeepAliveCount: 10,
            requestedPublishingInterval: 1000
        });

        subscription.on("raw_notification", (n) => {
            console.log(n.toString());
        });

        const itemsToMonitor = [
            {
                attributeId: opcua.AttributeIds.Value,
                nodeId: resolveNodeId("ns=3;s=AirConditioner_1.Temperature")
            },

            {
                attributeId: opcua.AttributeIds.Value,
                nodeId: resolveNodeId("ns=3;s=AirConditioner_2.Temperature")
            },
            {
                attributeId: opcua.AttributeIds.Value,
                nodeId: resolveNodeId("ns=3;s=AirConditioner_3.Temperature")
            },
            {
                attributeId: opcua.AttributeIds.Value,
                nodeId: resolveNodeId("ns=3;s=AirConditioner_4.Temperature")
            },
           {
                attributeId: opcua.AttributeIds.Value,
                nodeId: resolveNodeId("ns=3;s=AirConditioner_5.Temperature")
            }
 
        ];

        const optionsGroup = {
            discardOldest: true,
            queueSize: 1,
            samplingInterval: 10
        };

        const monitoredItemGroup = opcua.ClientMonitoredItemGroup.create(subscription, itemsToMonitor, optionsGroup, opcua.TimestampsToReturn.Both);

        // subscription.on("item_added",function(monitoredItem){
        monitoredItemGroup.on("initialized",  () => {
            console.log(" Initialized !");
        });

        monitoredItemGroup.on("changed",  (monitoredItem, dataValue, index) => {
            console.log("Changed on ", index ,  dataValue.value.toString());
        });

        await new Promise((resolve) => setTimeout(resolve, 1000000));

        await monitoredItemGroup.terminate();

        await session.close();
        await client.disconnect();
        console.log("Done !");

    } catch (err) {
        console.log("Error", err);
    }
})();

```

manjushreenimbalkar, 2019-08-08:
Hello,

I tried the above code and it gives me this error. It is not even able to connect to the client.
18:37:42.779Z :message_builder_base          :108   Error   client0 cannot construct object with nodeID ns=0;i=449

manjushreenimbalkar, 2019-08-08:
What is the best way of reconnecting to opcua. The current code is not working for me.

erossignon, 2019-08-09:
You should not need to do anything about reconnecting, the node-opcua client takes care of this for you. 
I would requalify your request as a support request 
(x) support request => Please do not submit support request here, instead see use
    contact Sterfive for commercial support (contact@sterfive.com)

closed 2020-08-15 by stale[bot]

wakeupbits, 2021-03-28:
Hey,

I have also same issue.is there anyone who got solution for it

dark-matter-matters, 2022-04-27:
I had the same issue, the reconnection of the session is never fired. 
The only solution is to terminate the session and its subscription when the client is disconnected, and then re-create the session and the subscriptions when it's back online.   

```ts
this.client!.on("connection_lost", () => {
      // session.on("session_restored") does not work. We have to destroy the session
      // and the current subscriptions and re-create them when we're back online.
      if (this.session) {
        this.session.close(true);
      }
      if (this.subscription && this.subscription.isActive) {
        this.subscription!.removeAllListeners();
        this.subscription!.terminate();
      }
    });

this.client!.on("after_reconnection", async () => {
      // Re-create session and subscriptions
      this.session = await this.client!.createSession();
      this.monitorNodesById(); // creates a subscrption and a ClientMonitoredItemGroup
    });
```