node-opcua/node-opcua #641: Error initializing field for LocalizedText

msalafia, 2019-07-11:
- (x) bug report
 - ( ) feature request
 - ( ) support request

**Current behavior**
I am just running a simple project created using 2.0.0 version of node-opcua and it returns the following error:

[image]

With version 0.7.4 it returns the following error:

[image]

With version 0.6.0 everything is fine.

**Expected behavior**
Should run 

**Please tell us about your environment:**
Windows 10, Node.js 10.7.0

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (X)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
   
* **Node:** 
   `node --version` = 10.7.0

erossignon, 2019-07-11:
can you provide the code ?

msalafia, 2019-07-12:
There is no specific piece of code generating this error. You can reproduce this issue cloning [this](https://github.com/OPCUAUniCT/node-opcua-coreaas) repo. Then, change in package.json the dependency `node-opcua: ^2.0.0` and try to execute demo.js or demo2.js in demo folder. In this file, change `require("node-opcua-coreaas")(opcua);` with `require("../index")(opcua);` and then try to run with `node .\demo.js`.

erossignon, 2019-07-12:
It looks like you're mixing the new version and an old version of node-opcua
all your node-opcua modules in the package.json file must have the same version number.
Expectially "node-opcua-generator" still refers to a very old version.

msalafia, 2019-07-14:
Thank you for your help, but i have noticed that Enum is not exported by the node-opcua module anymore. Is there any new way to get the Enum class starting from the object `opcua = require("node-opcua")` without requiring the module node-opcua-enum?

erossignon, 2019-07-14:
In node-opcua@2.0.0 onward, standard typescript enum are used ...

msalafia, 2019-07-14:
Thanmkl you very much @erossignon. I am closing this issue.
Anyway i have to change lot of old stuff since API changes a lot for most of the internal stuff. In particular, since i generate my own StructuredType, they are not working anymore because functions like register_class_definition or resolve_schema_field_types don't extist anymore. I will try to understand the new strategy. If you can give me advises they will be appreciated. Let me know if you prefer a new issue for this.

closed 2019-07-14 by msalafia