node-opcua/node-opcua #644: Unable to connect to node opc server (V2.0 )with Basic256Sha256 & SignAndEncrypt using node opc client

samminen, 2019-07-11:
I am using simple_server.js with below security enabled. When I connect to this server using node opc client using "None" and "None", it works. But it not connecting when I use "Basic256Sha256" and SignAndEncrypt.

Server raising the error:  Sender Certificate Error BadCertificateUntrusted (0x801a0000)

But I am using same server certificates both client side and server side to connect to this node server. 

I faced same problem when I try to connect to my OPC server using UAExpert client as well. 

Does node OPC server support all OPC security policies and security mode ? 

*********************************************

securityPolicies: [
        opcua.SecurityPolicy.Basic256Sha256,
        opcua.SecurityPolicy.None
    ],

    securityModes: [
        opcua.MessageSecurityMode.SignAndEncrypt,
        opcua.MessageSecurityMode.None
    ],
*********************************************

msalafia, 2019-07-19:
Try to create the OPC UA server in this way:

```javascript
let server = new OPCUAServer({
    nodeset_filename: xmlFiles,
    port: 4848,
    serverCertificateManager: new OPCUACertificateManager({ 
        automaticallyAcceptUnknownCertificate: true,
        rootFolder: path.join(__dirname, "../certs")
    })
})
```

victorkostin13, 2019-09-04:
> Try to create the OPC UA server in this way:
> 
> ```js
> let server = new OPCUAServer({
>     nodeset_filename: xmlFiles,
>     port: 4848,
>     serverCertificateManager: new OPCUACertificateManager({ 
>         automaticallyAcceptUnknownCertificate: true,
>         rootFolder: path.join(__dirname, "../certs")
>     })
> })
> ```

still not working...

samminen, 2019-09-04:
Thank you; It resolved issue.

closed 2019-09-17 by erossignon