node-opcua/node-opcua #647: OPCUA Server does not start with alternateHostname

Camarbro, 2019-07-18:
**I'm submitting a ...**  (check one with "x")

 - (X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

When creating an OPC UA server and specifying in the server options the alternate hostname as a single option or as part of alternate endpoints, this option gets ignored and uses the computer's hostname.

Example:
`
const server = new opcua.OPCUAServer({
    resourcePath: "Sample",
    port: 4335,
    alternateHostname: "someotherhostname"     
});
// When started, the server opcua address looks loke opc.tcp://computerHostname:4335/Sample
`

**Expected behavior**

`
const server = new opcua.OPCUAServer({
    resourcePath: "Sample",
    port: 4335,
    alternateHostname: "someotherhostname"     
});
// Expected server with hostname opc.tcp://someotherHostname:4335/Sample
`

**Minimal reproduction of the problem with instructions**

Use the example code provided above and create a simple server
  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

Previously in version 0.7.0, this would work perfectly but since I update to version 2.1.0 it doesn't

**Please tell us about your environment:**

Node opcua server in Mojave

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( X)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Macbook 2013
  * OS version: Mojave 10.14
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

2.1.0

* **Node:** 
   `node --version` =10.15

erossignon, 2019-07-19:
By design alternate hostname add a alternate name to the default computer name, and doesn't replace it

Camarbro, 2019-07-26:
I understand that part, It's just that I can't really connect with the alternate hostname and the error of duplicate endpoints troubles me. 
Why is that?

erossignon, 2019-08-23:
this is generally used for opcua server that have 2 network card (belongs to two separated isolated network) that may have to be accessed from different ip
this may help https://unix.stackexchange.com/questions/500050/does-a-computer-system-linux-only-have-one-host-name

padraic-padraic, 2020-06-30:
I would like to bump this issue, which I have also run into while trying to use node-opcua as a server while running in a Docker container. In this instance, the hostname returned by `os.hostname()` is not meaningful, and changes on container restarts.

Whether this duplicate set of endpoint descriptions causes problems or not  seems to depend on the client. We have run into issues trying to connect with a National Instruments OPC-UA client with the Sign and Encrypt profile, which seems to use the provided hostname to connect and perform the GetEndpoints request, before checking the endpoint URL against the certificate hostname. Because the container's hostname is random, and is used for the endpointURL, it obviously does not match the certificate Common Name and so the connection fails. Creating the certificate on container startup is also not feasible as we want the client to verify the server certificate.

Looking through the source, I can see that it's actually possible to supply the hostname when calling the `AddStandardEndpointDescriptions` method of `OPCUAServerEndpoint` (https://github.com/node-opcua/node-opcua/blob/eeade1b9/packages/node-opcua-server/source/server_end_point.ts#L131). So one fix would be to expose hostname as a server option, defaulting to getFullyQualifiedDomainName as currently happens here https://github.com/node-opcua/node-opcua/blob/eeade1b9/packages/node-opcua-server/source/server_end_point.ts#L331

erossignon, 2020-06-30:
You can force hostname when running the docker container by using the `-h` options.

```
$docker run -it -h my_hostname1234 ubuntu /bin/bash -c "echo my hostname is: && hostname"
```

closed 2020-11-07 by erossignon