node-opcua/node-opcua #650: new opcua.Variant with datatype opcua.DataType.Int64 from a value of -2147483647 asserts

emr-anthony-amaro, 2019-07-29:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

```
var testVal = -1;
var test = new opcua.Variant({dataType: opcua.DataType.Int64, value: testVal});
```

will throw an assert:

```
assert (index.ts 12:21)
constructInt64 (integers.ts 177:5)
coerceUInt64 (integers.ts 208:12)
coerceVariantType (variant.ts 801:21)
constructHook (variant.ts 487:25)
Variant (variant.ts 96.19)
```

**Expected behavior**
Just looking at the constructInt64 function, it seems that any negative number coming in as an Int64 variant would cause this problem. I guess I would expect a signed value to work in an Int64 variant. Not trying to be snarky, stuff happens.

**Minimal reproduction of the problem with instructions**

The following three lines should produce the assert
```
var opcua = require("node-opcua");
var testVal = -1;
var test = new opcua.Variant({dataType: opcua.DataType.Int64, value: testVal});
```

**What is the motivation / use case for changing the behavior?**
Signed (negative) values should work with Variants of type Int64.

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( x)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: __MacBook Pro___
  * OS version: __Mojave 10.14.5___
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - (x ) MacOs   : version : ___10.14.5______
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:___N/A______
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

6.2.0

"gitHead": "ddce2b6e3e13371510e63d839c1a4c4ff0be2c15",

* **Node:** 
   `node --version` = v12.6.0

closed 2020-08-06 by stale[bot]

SebsSa, 2022-05-18:
Hello,
this bug does not seem to be fixed yet.
@erossignon could we get an update if it is being worked on ?

erossignon, 2022-05-19:
@[SebsSa](https://github.com/SebsSa) , if you would like to contribute helping us fixing the issue, please contact us so we can discuss how to prioritize this point.

dark-matter-matters, 2022-07-18:
I have noticed this problem, when I try to convert any negative value node-opcua throws an error : 
```log
Error occurred in handler for '/Class/change_snv': Error
     at assert (C:\Project\node_modules\node-opcua-assert\dist\index.js:11:21)
     at constructInt64 (C:\Project\node_modules\node-opcua-basic-types\dist\integers.js:163:36)
     at coerceUInt64 (C:\Project\node_modules\node-opcua-basic-types\dist\integers.js:184:16)
     at coerceVariantType (C:\Project\node_modules\node-opcua-variant\dist\variant.js:642:62)
     at constructHook (C:\Project\node_modules\node-opcua-variant\dist\variant.js:381:25)
     at new Variant (C:\Project\node_modules\node-opcua-variant\dist\variant.js:59:26)
     at new DataValue (C:\Project\node_modules\node-opcua-data-value\dist\datavalue.js:237:42)
     at C:\Project\build\server\opcua-electron\OpcUaFunctions.js:254:20
     at Array.forEach (<anonymous>)
     at C:\Project\build\server\opcua-electron\OpcUaFunctions.js:250:13
```
The exact same piece of code with DataType.Int32 or DataType.Int16 works as expected. The type of the value provided is `number`. I find it strange that `coerceUInt64`is called on the stack trace, but perhaps this is normal.

dark-matter-matters, 2022-08-22:
It turns out `Int64` implementation returns a two-dimmensions array which works quite unexpectedly (at least I couldn not find any reference to it in the documentation).  
When reading the value fromthe server, if the value is positive then we should just take the second value from the array. If it is negative, we have to substract the last element from the first one, multiply it by -1 and substract -1 to get the value.  
Here are the two helpers functons I wrote to use this data type : 
```typescript 
const INT64MAX = 4294967295; //2^32 - 1
export const getInt64ValueFromArray = (valueToDecode: Array<number>) => {
  if (!valueToDecode || !valueToDecode.length || valueToDecode.length < 2) {
    console.warn(
      "Invalid int64 array provided, need two-dimmensions array",
      valueToDecode
    );
    return valueToDecode;
  }
  if (valueToDecode[0] === 0) {
    return valueToDecode[1];
  } else {
    return -valueToDecode[0] + valueToDecode[1] - 1;
  }
};

export const convertInt64ToArray = (valueToEncode: number) => {
  if (valueToEncode >= 0) {
    return [0, valueToEncode];
  } else {
    return [INT64MAX, INT64MAX - valueToEncode];
  }
};
```

erossignon, 2022-08-22:
@dark-matter-matters  as you figured out , int64 are handled as 2xInt32 inside an array.

convertInt64ToArray  => you can use coerceInt64 or coerceUInt64 to convert number to  NodeOPCUA Int64 proxy arrays.

getInt64ValueFromArray  has no equivalent in node-opcua; beware that the method you are proposing may be loosing accuracy.

The reason for this is that nodejs and javascript do not support int64 natively, although since node >=14 we could use BigInt.
we still have a customer using nodejs 12 which prevents us to migrate immediately into using Int64.

We will deprecate the use of nodeJS 12 and be in a position to use BigInt instead.
The work is not planned anytime soon, and we are expecting sponsors to prioritize the transition into using BigInt to store Int64 & UInt64 in node-opcua.

If your company is interested to help,  please contact us so we can arrange something.

dark-matter-matters, 2022-08-23:
Thanks for the answer, now I understand better how it works. 
There is still something that I don't understand with negative values. When calling `coerceInt64()` with a negative value, an error is thrown (the one mentionned [in my first reply](https://github.com/node-opcua/node-opcua/issues/650#issuecomment-1187296391)). There seems to be a link to `convertUInt64()`, and it returns a *UInt64* instead of a *Int64* : 
```typescript
export declare function coerceUInt64(value: number | UInt64 | Int32 | string | null): UInt64;
export declare function randomInt64(): Int64;
export declare const coerceInt64: typeof coerceUInt64;
```