node-opcua/node-opcua #651: Source and/or Server Timestamp do not reflect current machine time

eg-be, 2019-07-30:
**I'm submitting a ...**  (check one with "x")

 - ( x ) bug report

**Current behavior**
If the System time changes, this has no effect on the Source- and / or Server timestamp generated by node-opcua.

It seems: Opcua (or node?) reads the current system time upon creating the server (starting node?) and then calculates all later timestamp from that value on(?)

**Expected behavior**
That Source- and Server timestamp values reflect the timestamp of the machine where the OpcuaServer is running on.
A real use-case I observed is that the machine time gets updates from NTP - this makes the opcua-server timestamps slowly drift away from the actual system time on the machine.

**Minimal reproduction of the problem with instructions**
Start the Sample-Server demo included in the source of node-opcua ( https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_server.js ). Connect to the Server using some opcua-client (UAExpert for example). Update the system-time of the machine where the Server is running on. The Source- and Server-Timestamp of the items will not reflect this change in system time.

**What is the motivation / use case for changing the behavior?**
Updates from NTP.

**Please tell us about your environment:**
  * ( x )  my request is related to node-opcua acting as a OPCUA SERVER 
  * ( x ) I have installed node-opcua as a package ( using npm install )
  * ( x ) Windows : version : 7 Professional    
    
* **node-opcua version:**  :
`-- node-opcua@2.1.0

* **Node:** 
   `node --version` =v10.16.0

[image]

erossignon, 2019-07-31:
Good catch:
This is because sourceTimestamp/serverTimestamp use a higher precision (and also faster to compute) clock than just the plain old javascript  ````new Date();``` a.k.a ````process.hrtime(origin)`````.
 
A fix would be to resync the origin value now and then.

see https://github.com/node-opcua/node-opcua/blob/2c8d07ac1a3b6cae03884e7727ceaae98488b215/packages/node-opcua-date-time/source/date_time.ts#L168

erossignon, 2019-08-18:
This has been fixed in node-opcua@2.1.5  by resyncing getCurrentClock  with system clock every 30 seconds. @eg-be, please confirm

eg-be, 2019-08-22:
@erossignon I can confirm that the fix in node-opcua@2.1.5 resolves the problem. Thank you!

closed 2019-08-23 by erossignon

Akeel95, 2023-12-12:
Hi @erossignon ,

I am using the latest version (2.118.0) of node-opcua with [this](https://github.com/node-opcua/node-opcua-sampleserver) sample server.
The sourceTimestamp and/or serverTimestamp are not updating for me.
But the same is working for the node-opcua version 2.75.0.
I attached the image below.

Image for V 2.118.0 :
[image]

Image for V 2.75.0 :
[image]

Would you please check and update?

Thanks for the awesome lib !!

mb-software, 2024-02-23:
> Hi @erossignon ,
> 
> I am using the latest version (2.118.0) of node-opcua with [this](https://github.com/node-opcua/node-opcua-sampleserver) sample server. The sourceTimestamp and/or serverTimestamp are not updating for me. But the same is working for the node-opcua version 2.75.0. I attached the image below.
> 
> Image for V 2.118.0 : ![node-opcua-2 118 0](https://private-user-images.githubusercontent.com/153627817/289892395-62892c39-8442-440f-a9ae-6bdf9e704afd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2OTg5MDgsIm5iZiI6MTcwODY5ODYwOCwicGF0aCI6Ii8xNTM2Mjc4MTcvMjg5ODkyMzk1LTYyODkyYzM5LTg0NDItNDQwZi1hOWFlLTZiZGY5ZTcwNGFmZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyM1QxNDMwMDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZTM4OTJlYzY5OTYyZWE1ZTIyYjY4NjRlODVlOGI5MjhkM2I1ZmU1ZWRmZmMwY2JlZmU5ZmRiOGY3ZjQ5NTgxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1AMWz1Js_XUHO8Db4_5oHXGvvTPjHG6-RQYyXqNlC9I)
> 
> Image for V 2.75.0 : ![node-opcua-2 75 0](https://private-user-images.githubusercontent.com/153627817/289892453-8745aa0b-6886-426b-ab81-6afec98cdb39.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2OTg5MDgsIm5iZiI6MTcwODY5ODYwOCwicGF0aCI6Ii8xNTM2Mjc4MTcvMjg5ODkyNDUzLTg3NDVhYTBiLTY4ODYtNDI2Yi1hYjgxLTZhZmVjOThjZGIzOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyM1QxNDMwMDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNzA5ZDk3NzkyODM5YjAyNjY4YTJjYTZlZmE0MTBiODAwNTM2MWI2ZTFjMWQxMzZkZTNmNDUzNWMzYTlhZDNiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.brfbZE89a7QKuMZfgmhgVzN1JANVBmNt7gYS-gTzjQY)
> 
> Would you please check and update?
> 
> Thanks for the awesome lib !!

Hi,
I confirm I have the same behaviour in latest version (now 2.120.0).
I am surprised no one else noticed this? 
Should we open a new issue about it?