node-opcua/node-opcua #652: unable to get node node values from opcua server using node js

kumarkumaresh, 2019-08-02:
i am connected to opcua server using node js code.
i get node ids perfectly using UA Expert tool but when i am trying to get node values from node js code it throws invalid node id error.
ua expert
=======
[image]

code
===
[image]

error
====
[image]

erossignon, 2019-08-02:
it looks like the nodeId you want to access are string based... but they also contain quotes (")

I would replace 
`"ns=3;s=SomeText"` with  "ns=3;s=\"SomeText\""`

erossignon, 2019-08-02:
duplicates #623

closed 2019-08-02 by erossignon

kumarkumaresh, 2019-08-05:
@erossignon Thanks for your reply, yes it worked.
thanks a lot again.