node-opcua/node-opcua #655: crawler is not reading all the elements of the array if the array size is greater than 100'

Nandish-Reddy, 2019-08-07:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =

erossignon, 2019-08-07:
Can you descrive hte other OPCUA system you are trying to connect to ?
May this server is not supporting large browse and read Request ... and is not exposing its limit the way it should.

I would suggest that you limit the nodeCrawler 

```
        const crawler = new NodeCrawler(session);
        crawler.maxNodesPerBrowse = 50;
        crawler.maxNodesPerRead = 50;
``

Nandish-Reddy, 2019-08-09:
Hi Etienne,

Thanks for looking into this.

Tried the above configurations but it did not help us. 

Python opcua gui client and UAexpert both were able to browse all the elements of large array.

The data type field for array node is not properly set,which is undefined.

while i browse a node arr[0..199] using crawler, it browses from 100th element, 101,102 ...199.

One more observation, In the packets seen in wireshark , i am able to see all the array elements data. which has reached node opcua client from opcua server system.

I am using UA opcua server.

erossignon, 2019-08-09:
Please provide source code that you are using to access  the server so issue can be reproduced

jagadishpb89, 2019-08-09:
Hi Etienne, 

Thanks for quick reply on this. I was also looking for this.

I am also facing the same issue for even scalar data types. When I have scalar data type elements with names IntVar1, IntVar2, IntVar3,...........IntVar1000 in opcua server, it is printing all the elements starting from IntVar101 to IntVar1000. Missed elements from IntVar1 to IntVar100.

We are not getting 'browsed' events for remaining elements.

If I have only 98 elements from IntVar1 to IntVar97, then node-crawler browses everything, we getall the 'browsed' events. 

Please find below the crawler code.
node-opcua version: 2.0.0

              var opcua = require("node-opcua");
               var crawler = new opcua.NodeCrawler(the_session);
               var nodeId="ObjectsFolder"
               console.log("Browsing data in the server");
               crawler.read(nodeId, function (err, obj) {
                  if (err) {
                     console.log("Failed to Read Data");
                  }
                  else {
                     console.log("data read from PLC is Successful");
                  }
                  cb3(err);
               });
               crawler.on("browsed", function (element) {
               }

You can add some scalar data type elements in any opcua server, try to browse as per above code. You may reproduce the issue.

Thanks

erossignon, 2019-08-09:
the code looks incomplete. could you provide an complete standalone example  ?

jagadishpb89, 2019-08-09:
Here is complete code.

async.series([
   function (cb3) {

      client.connect(endpointUrl, function (err) {

         if (err) {
            console.log(" cannot connect to endpoint :", endpointUrl);
         } else {
            console.log("connected !");
         }
         cb3(err);

      });

   },
   function (cb3) {
      client.createSession(function (err, session) {
         if (err) {
            console.log("Failed to Create the Browse Session");
         }
         else {
            console.log("Browse Session Creates Session Successfully!");
            the_session = session;
         }
         cb3(err);
      });

   },
   function (cb3) {
      var crawler = new opcua.NodeCrawler(the_session);
      var nodeId = "ns=3;i=85"
      console.log("Browsing Data in the server");
      crawler.read(nodeId, function (err, obj) {
         if (err) {
            console.log("Failed to Read Data");
         }
         else {
            console.log("Data read from PLC is Successful");
         }
         cb3(err);
      });

      crawler.on("browsed", function (element) {
         console.log("element name:" + element.browseName.name);
      });
   }
],
   function (err) {
      if (err) {
         console.log("Failure in opcua browsing!", err);
         callback(err)
      } else {
         console.log("No Error :)");
         callback()
      }
      console.log("disconnected");
      client.disconnect(function () { });
   })

erossignon, 2019-08-10:
@jagadishpb89 running your sample on the Unified Automation Ansi C Server provide the following result on my end . as you can see all node of the Mass Variable foder are properly extracted.

see this [gist](https://gist.github.com/erossignon/84e9ec090909da037401d4c29b51b85b)

what is the result on your end ? ( please provide a full gist)

erossignon, 2019-08-11:
I have modified  NodeCrawler so that when  `crawler.maxNodesPerBrowse` or crawler.maxNodesPerRead` is set,  the value serves as a minimum that can restrict the value provided by the server
```javascript
const crawler = new NodeCrawler(session);
crawler.maxNodesPerBrowse = 50;
crawler.maxNodesPerRead = 50;
```
this is available in node-opcua@2.1.3

jagadishpb89, 2019-08-13:
We have tried with above configurations. Unfortunately we still could not get all the nodes.

Looked into the node crawler code, applied some debug logs there. I did print the browse results in browse function and browseNext functions .

My opcua server had 600 elements , first 100 references of a node  were getting printed as part of browse() response itself. Next 500 references were being printed as part of browseNext()'s response. where each response were having 100 elements , i got 5 such browseNext's responses with continuationPoints results with 100 elements each.

Now , I am missing browse emit events for the elements (first 100) which browse() api has returned.

We are still debugging the code, Thanks a lot for the update.

Nandish-Reddy, 2019-08-13:
> We have tried with above configurations. Unfortunately we still could not get all the nodes.
>
> Looked into the node crawler code, applied some debug logs there. I did print the browse results in [...]

 Facing similar issue.In some opcua servers, i see first 1000 elements are skipped  if elements are more than 1000. If elements are less than 1000 it prints all the elements.

erossignon, 2019-08-15:
@Nandish-Reddy ; there is nothing we can do with statement like "with some opcua servers" , as we need to be able to reproduce the problem for investigation.

please answer the question asked above, and that are left unanswered in your reply so far or not accurately enough:

  -Can you describe the other OPCUA system you are trying to connect to ?
    OPCUA server vendor, version number, etc ...

  - Please provide source code of the client program (js or ts + package.json etc ) that you are using to access the server so issue can be reproduced. Please provide minimal , yet complete functional code

Also I am adding 2 more questions for you:

   - have you try upgrading to node-opcua@2.1.3
  - what is the output of the [sample program I have provided in this gist](https://gist.github.com/erossignon/84e9ec090909da037401d4c29b51b85b)

erossignon, 2019-08-16:
OK.. I can now reproduce it, it happens when the server limits the number of browseResult and provides a continuation point ...

erossignon, 2019-08-18:
should be fixed now in version node-opcua@2.1.5 , @Nandish-Reddy , @jagadishpb89 please confirm.

jagadishpb89, 2019-08-19:
Hi Etienne,

Thanks a lot for your great help. Sorry for late reply.
I was debugging the issue, thinking it is something to do with synchronous read and browse operations.

I tried with version node-opcua@2.1.5. It works. You can close the issue.

closed 2019-08-19 by erossignon

binaryPUNCH, 2019-10-01:
´How do you guys create NodeCrawlerClientSession to begin with?