node-opcua/node-opcua #658: dataValue.value.dataType returns only ids

mariopoeta, 2019-08-12:
Hi,

I am currently using nodeopcua 2.1.3, and when i am trying to retrieve datatype it returns ids, like 11 (which is dobule). How can I retrieve "double" like in 0.0.65 for example?

mariopoeta, 2019-08-12:
code:        	

const monitoredItems = the_subscription.monitorItems(itemsToMonitor,requestedParameters,opcua.TimestampsToReturn.Both).then(function(monitoredItems) {
       		monitoredItems.on("changed", function (item, dataValue, index) {
                var payload = {
                    Data: {
                        DataType: dataValue.value.dataType,
                        CollectionTimestamp: new Date().toISOString(),
                        Value:
                        {
                            Value: dataValue.value.value,
                            StatusCode: dataValue.statusCode.name,
                            SourceTimestamp: dataValue.sourceTimestamp.toISOString(),
                            ServerTimestamp: dataValue.serverTimestamp.toISOString(),
                        },
                    },
				};
				console.log(payload);
       		}); 
       	});

mariopoeta, 2019-08-12:
{ Data:

   { DataType: 11,

     CollectionTimestamp: '2019-08-12T13:23:04.296Z',

     Value:

      { Value: 10071.374,

        StatusCode: 'Good',

        SourceTimestamp: '2019-08-12T13:23:03.784Z',

        ServerTimestamp: '2019-08-12T13:23:03.784Z' } } }

{ typeOfDataType:

      { '0': 'Null',

        '1': 'Boolean',

        '2': 'SByte',

        '3': 'Byte',

        '4': 'Int16',

        '5': 'UInt16',

        '6': 'Int32',

        '7': 'UInt32',

        '8': 'Int64',

        '9': 'UInt64',

        '10': 'Float',

        '11': 'Double',

        '12': 'String',

        '13': 'DateTime',

        '14': 'Guid',

        '15': 'ByteString',

        '16': 'XmlElement',

        '17': 'NodeId',

        '18': 'ExpandedNodeId',

        '19': 'StatusCode',

        '20': 'QualifiedName',

        '21': 'LocalizedText',

        '22': 'ExtensionObject',

        '23': 'DataValue',

        '24': 'Variant',

        '25': 'DiagnosticInfo' },

So i want the Double, which is 11.

erossignon, 2019-08-15:
you could try
```
import { DataType } from "node-opcua";

console.log("DataType in string form",DataType[payload.dataType]);
```

mariopoeta, 2019-08-16:
Thx! i will check that!

mariopoeta, 2019-08-16:
[2019-08-16T11:53:56.932Z][ERROR]-import { DataType } from "node-opcua";
[2019-08-16T11:53:56.932Z][ERROR]-^^^^^^
[2019-08-16T11:53:56.932Z][ERROR]-SyntaxError: Unexpected token import
[2019-08-16T11:53:56.932Z][ERROR]-    at createScript (vm.js:80:10)
[2019-08-16T11:53:56.932Z][ERROR]-    at Object.runInThisContext (vm.js:139:10)
[2019-08-16T11:53:56.932Z][ERROR]-    at Module._compile (module.js:617:28)
[2019-08-16T11:53:56.932Z][ERROR]-    at Object.Module._extensions..js (module.js:664:10)
[2019-08-16T11:53:56.932Z][ERROR]-    at Module.load (module.js:566:32)
[2019-08-16T11:53:56.932Z][ERROR]-    at tryModuleLoad (module.js:506:12)
[2019-08-16T11:53:56.932Z][ERROR]-    at Function.Module._load (module.js:498:3)
[2019-08-16T11:53:56.932Z][ERROR]-    at Module.require (module.js:597:17)
[2019-08-16T11:53:56.932Z][ERROR]-    at require (internal/module.js:11:18)
[2019-08-16T11:53:56.932Z][ERROR]-    at new Runtime (/runtime/nodejs/lambda_nodejs_runtime.js:210:28)

mariopoeta, 2019-08-16:
However, already have a workaround for this.

erossignon, 2019-08-17:
`import { DataType } from "node-opcua"; ` should go at the top of your file if you are using typescript
and should be `const { DataType } = require("node-opcua")` if your script is in javascript.

closed 2019-08-17 by erossignon