node-opcua/node-opcua #660: BadNodeIdUnknown (0x80340000) but UA Expert can read node values from SIMETIC-ET200sp

mahboobelahi93, 2019-08-13:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - (x ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

I am reading i/o status from Et-200sp using OPC and using node-opcua API but client application can not read and gives the statusCode **BadNodeIdUnknown (0x80340000) but i can read node values from UA Expert......
Also when i tried to write a value to the node then description attribute says "There is no node in server namespace" so how can i fix this issue** 

**Expected behavior**

Apparently, there is no bug in code.
**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**
Windows,opc client,node-opcua

  * (x ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( x) I have installed node-opcua from source ( using git clone)
  * ( x) I have installed node-opcua as a package ( using npm install )
  * ( x) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - (x ) other : please specify 
        
  * Device: __ET200sp___
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =

allatrue, 2019-08-13:
Maybe NodeId or Namespace is wrong, as error message says? Which node you are trying to read or write?

mahboobelahi93, 2019-08-13:
I took nodeid( namespace and I'd ) from ua expert which read the value of each node correctly.... I want to read ns=3;s="start".

allatrue, 2019-08-13:
Post your code and UaExpert screenshot

mahboobelahi93, 2019-08-13:
`import {
    OPCUAClient,
    MessageSecurityMode, SecurityPolicy,
    AttributeIds,
    makeBrowsePath,
    ClientSubscription,
    TimestampsToReturn,
    MonitoringParametersOptions,
    ReadValueIdLike,
    ClientMonitoredItem,
    DataValue
} from "node-opcua";

const connectionStrategy = {
    initialDelay: 1000,
    maxRetry: 1
}
const options = {
    applicationName: "Simatic_ET200sp_Client",
    connectionStrategy: connectionStrategy,
    securityMode: MessageSecurityMode.None,
    securityPolicy: SecurityPolicy.None,
    endpoint_must_exist: false,
};
const client = OPCUAClient.create(options);
// const endpointUrl = "opc.tcp://opcuademo.sterfive.com:26543";
const endpointUrl = "opc.tcp://10.1.1.12:4840";
async function main() {
    try {
        // step 1 : connect to
        await client.connect(endpointUrl);
        console.log("connected !");

        // step 2 : createSession
        const session = await client.createSession();
        console.log("session created !");

        // step 3 : browse
        const browseResult = await session.browse("RootFolder");

        console.log("references of RootFolder :");
        for (const reference of browseResult.references) {
            console.log("   -> ", reference.browseName.toString());
        }

        // step 4 : read a variable with readVariableValue
        //ns=0;i=2261 value: SIMATIC S7-1500 OPC UA
        const dataValue2 = await session.readVariableValue('ns=3;s=start');//ns=3;s=start
        console.log(" START = ", dataValue2.toString());

        // step 4' : read a variable with read
        const maxAge = 0;
        const nodeToRead = {
            nodeId: "ns=3;s=motor",
            attributeId: AttributeIds.Value
        };
        const dataValue = await session.read(nodeToRead, maxAge);
        console.log(" MOTOR ", dataValue.toString());

        // step 5: install a subscription and install a monitored item for 10 seconds
        const subscription = ClientSubscription.create(session, {
            requestedPublishingInterval: 1000,
            requestedLifetimeCount: 100,
            requestedMaxKeepAliveCount: 10,
            maxNotificationsPerPublish: 100,
            publishingEnabled: true,
            priority: 10
        });

        subscription.on("started", function () {
            console.log("subscription started for 2 seconds - subscriptionId=", subscription.subscriptionId);
        }).on("keepalive", function () {
            console.log("keepalive");
        }).on("terminated", function () {
            console.log("terminated");
        });

        // install monitored item

        const itemToMonitor: ReadValueIdLike = {
            nodeId: "ns=3;s=motor",
            attributeId: AttributeIds.Value
        };
        const parameters: MonitoringParametersOptions = {
            samplingInterval: 100,
            discardOldest: true,
            queueSize: 10
        };

        const monitoredItem = ClientMonitoredItem.create(
            subscription,
            itemToMonitor,
            parameters,
            TimestampsToReturn.Both
        );

        monitoredItem.on("changed", (dataValue: DataValue) => {
            console.log(" value has changed : ", dataValue.value.toString());
        });

        /*async function timeout(ms: number) {
            return new Promise(resolve => setTimeout(resolve, ms));
        }
        await timeout(10000);

        console.log("now terminating subscription");
        await subscription.terminate();*/

        // step 6: finding the nodeId of a node by Browse name   Server.ServerStatus.BuildInfo.ProductName
        const browsePath = makeBrowsePath("RootFolder", "/Objects/Server.ServerStatus.BuildInfo.ProductName");

        const result = await session.translateBrowsePath(browsePath);
        //console.log('RESULT====>>>>', result);

        const productNameNodeId = result.targets[0].targetId;
        console.log(" Product Name nodeId = ", productNameNodeId.toString());

        // close session
        await session.close();

        // disconnecting
        await client.disconnect();
        console.log("done !");
    } catch (err) {
        console.log("An error has occured : ", err);
    }
}
main();`
[image]
[image]

mahboobelahi93, 2019-08-13:
monitored node ids:
ns=3;s=start
ns=3;s=motor

allatrue, 2019-08-13:
See #652. Maybe it's also your case?

mahboobelahi93, 2019-08-13:
ALLATRUE thanks a lot it works fine

closed 2019-08-15 by erossignon