node-opcua/node-opcua #662: @erossignon erossignon Hi Etienne I am using the following code for writing node value but it seems that  **await session.write(nodeToWrite);** does not eork correctly also i am attaching pic of error.

mahboobelahi93, 2019-08-15:
@erossignon erossignon Hi Etienne I am using the following code for writing node value but it seems that  **await session.write(nodeToWrite);** does not eork correctly also i am attaching pic of error.
`        let val = await session.read(nodeIds[0])
        console.log(val.value.toString());

        let nodeToWrite = {
            nodeId: nodeIds[0],// 'ns=3;s="INPUT_1"'
            attributeId: opcua.AttributeIds.Value,
            indexRange: null,
            value: /* DataValue */ {
                //sourceTimestamp: new Date(),
                statusCode: opcua.StatusCodes.Good,// <==== 
                value: /* Variant */ {
                    dataType: opcua.DataType.Boolean,
                    value: true
                }
            },
        }
        let risultato = await session.write(nodeToWrite);
        console.log(risultato.toString());

        val = await session.read(nodeIds[0])
        console.log(val.value.toString())`

[image]

_Originally posted by @mahboobelahi93 in https://github.com/node-opcua/node-opcua/issues/462#issuecomment-521634386_

erossignon, 2019-08-15:
This is a known issue in the typescript definition file that requires you to specify a complete dataValue.
` value:  new DataValue({...})` instead of value: {...} 
It should be solved this way, as a workaround:
``` javascript
let nodeToWrite = {
        nodeId: nodeIds[0],// 'ns=3;s="INPUT_1"'
        attributeId: opcua.AttributeIds.Value,
        indexRange: null,
        value:  new DataValue({       //<<<< PLEASE ADD THIS
            //sourceTimestamp: new Date(),
            statusCode: opcua.StatusCodes.Good,// <==== 
            value: /* Variant */ {
                dataType: opcua.DataType.Boolean,
                value: true
            })
        },
    }
``

mahboobelahi93, 2019-08-17:
@erossignon erossignon Hi Etienne now value to the node written is successful but it does not change previous value.
Actually, i am trying to write value to Simens **Et200sp (CPU1512SP F-1PN)** input from opcua client but the written value has no effect on the previous one as shown in figure also i can't able to write from UA Expert.
code segment:
//TO WRITE
`        console.log('\nWriting starts_L91:\n');
        let val = await session.readVariableValue(nodeIds[0])
        console.log('Before Writing_L93: ', val.value.toString());

        let nodeToWrite = {
            nodeId: nodeIds[0],
            attributeId: opcua.AttributeIds.Value,
            indexRange: null,
            value: new DataValue({
                statusCode: opcua.StatusCodes.Good,
                value: {//Variant
                    dataType: opcua.DataType.Boolean,
                    value: true
                }
            })

        }
        let risultato = await session.write(nodeToWrite);
        console.log('Written Value_L109: ', risultato.toString());

        val = await session.readVariableValue(nodeIds[0])
        console.log('After writing_L112: ', val.value.toString())
        console.log('Status Code_L113: ', val.statusCode);
`
[image]

erossignon, 2019-08-18:
May be the value is not "writeable" on the server side...
Can you try this code to extract the accessLevel and userAccessLevel of the variable.

```javascript
const  nodesToRead = [
{ nodeId: nodeIds[0],     attributeId: AttributeIds.AccessLevel },
{ nodeId: nodeIds[0],     attributeId: AttributeIds.UserAccessLevel },
];
const dataValues = async session.read(nodesToRead);
console.log("      accessLevel =",dataValues[0].toString());
console.log(" useraccessLevel =",dataValues[1].toString());

```

mahboobelahi93, 2019-08-18:
Thanks. I will try in some time and come to back with latest information

mahboobelahi93, 2019-08-20:
@erossignon erossignon i have tried the above code and it seems that value is writeable
[image]

erossignon, 2019-08-20:
my hypothesis :
may the server claims that the variable is writable but is not.
may be the value is driven by an external process from within the server
may be the type of the value to write is not "Boolean"
may be there is a bug on the server side
Please contact the server manufacturer to find out.

alternativelly , you could try to remove the line indexRange: null in the node to write section.

mahboobelahi93, 2019-08-21:
Etienne please check this issue **#665**step 6: finding the nodeId of a node by Browse name

closed 2020-03-20 by erossignon