node-opcua/node-opcua #664: Write Array

wilclefison, 2019-08-20:
Hi....

I'm try write an array in tag but i don't know why i got Error:

My array = [ 2, 23, 23, 12, 24, 3, 25, 3, 26, 3, 27, 3, 28, 1, 43690, 1, 1261, 0, 0, 0, 0, 0, 0, 0, 65535, 11 ]

Conversion to Uint16

valor = new Uint16Array([ 2, 23, 23, 12, 24, 3, 25, 3, 26, 3, 27, 3, 28, 1, 43690, 1, 1261, 0, 0, 0, 0, 0, 0, 0, 65535, 11 ])                        
   

 nodeToWrite[Controlador] = {
                        nodeId:  resolveNodeId("ns=2;s=" + endereco[0].ADDRESS),
                        attributeId: opcua.AttributeIds.Value,
                        indexRange: null,
                        value: /*new DataValue(*/{
                            
                            value: {/* Variant */
                                dataType: tipoDado(parseInt(endereco[0].DATA_TYPE)),
                                arrayType: arrayTipo,
                                value: valor,
                            }
                        }
                    }

Result:

ConstantStatusCode {
    _value: 2147483648,
    _description: 'The value is bad but no specific reason is known.',
    _name: 'Bad' } ]

wilclefison, 2019-08-20:
Ohter error when i'm write array: Warning ! an array containing  UInt16 elements has been provided as a generic array. 
          This is inefficient as every array value will have to be coerced and verified against the expected type
          It is highly recommended that you use a  typed array  Function  instead

erossignon, 2019-08-23:
also covered by https://stackoverflow.com/questions/57585499/write-array-in-nodeopcua/57600250#57600250 , so closing

closed 2019-08-23 by erossignon