node-opcua/node-opcua #665: step 6: finding the nodeId of a node by Browse name

mahboobelahi93, 2019-08-20:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

@erossignon erossignon Hi Etienne I am trying to get nodeId's using brows name  and my folder hierarchy is shown in the following figure
[image]
got the following  error

[image]

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Windos_____
  * OS version: _____
    
    - ( ) Windows : version :10 _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:ET200sp
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =latest

mahboobelahi93, 2019-08-22:
Is anybody here to address this issue

erossignon, 2019-08-23:
please refer to Release 1.04 page 162 OPC Unified Architecture, Part 4 for BNF definition of RelativePath

mahboobelahi93, 2019-08-26:
Hi Etienne, i have checked BNF definition of Relative Path in  OPC Unified Architecture  Release 1.04 and it does not even take the following string for making a  relative path
`const browsePath = makeBrowsePath("RootFolder", "/Objects/DeviceSet");` 
but accept this one
`const browsePath = makeBrowsePath("RootFolder", "/Objects/Server");`
also "**

> DeviceSet, Processing and Server

**" all have same subtype of HierarchicalReferences

mahboobelahi93, 2019-08-26:
[image]
can you please also explains the difference between 1 and 2 in figure

erossignon, 2019-09-17:
DeviceSet do not belongs to namespace 0 and shall be preprending with namespace index in the path:

Such as const browsePath = makeBrowsePath("RootFolder", `/Objects/${nsDI}:DeviceSet");

closed 2020-03-20 by erossignon