node-opcua/node-opcua #666: The Typescript compiler throws multiple errors if "strictNullChecks" is set to true

Alia5, 2019-08-23:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
The Typescript compiler throws multiple errors if "strictNullChecks" is set to true in the **parent** project. (So node-opcua, is used as dependency)

**Expected behavior**

**Minimal reproduction of the problem with instructions**
1. Create new Typescript Project
2. npm i node-opcua --save-dev
3. add ```"strictNullChecks": true``` to compilerOptions of tsconfig.json (maybe strict: true as well...)
4. run tsc...
5. Lots of errors...

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * (x ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` = 2.1.5

erossignon, 2019-08-23:
please provide complete minimal project to reproduce
package.json, tsconfig , sample.ts file etc...

erossignon, 2019-08-23:
Step 5. Lots of errors... doesn't help

Alia5, 2019-08-26:
reproduce the steps, sigh...

erossignon, 2019-08-26:
Step 1 need to be detailed as well,

erossignon, 2019-10-19:
Please check `tsconfig.json` file is setup for modern javascript

closed 2019-10-19 by erossignon