node-opcua/node-opcua #669: Monitored Items with subscription.monitor throw errors

netbymatt, 2019-09-10:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
Following the notes in https://github.com/node-opcua/node-opcua/wiki/2.0.0-breaking-changes I've successfully created a client, session and subscription. When I get to the add creating monitored item section I can get variant 1 to work without any issues. But both the variant 2/callback and variant2/async (my preferred method) throws the following error:
Error: expecting a function here at Object.assert (/home/matt/cce/node_modules/node-opcua-assert/dist/index.js:10:21) at ClientSubscriptionImpl.monitor (/home/matt/cce/node_modules/node-opcua-client/dist/private/client_subscription_impl.js:147:29) at eval (eval at thenify.withCallback (/home/matt/cce/node_modules/thenify/index.js:33:10), :12:12) at new Promise () at ClientSubscriptionImpl.monitor (eval at thenify.withCallback (/home/matt/cce/node_modules/thenify/index.js:33:10), :10:8) at addMonitoredItem (/home/matt/cce/cce-opcua/monitored.js:30:35) at process._tickCallback (internal/process/next_tick.js:68:7)

**Expected behavior**
All three variants should produce the same results.

**Minimal reproduction of the problem with instructions**
See current behavior above
  
**What is the motivation / use case for changing the behavior?**
For consistency with code elsewhere in my project I would like to use the async/await syntax.

**Please tell us about your environment:**

  * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : Ubuntu 18.04
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:Siemens S7-1500
     - Version:V2.1
    
* **node-opcua version:**  : 2.1.7

* **Node:** 
   `node --version` =10.15.3

erossignon, 2019-09-13:
Please provide code extract for investigation

netbymatt, 2019-09-16:
This code works:
```
const addMonitoredItem = async (tagUnescaped, callback) => {
	// escape the tag name
	const tag = utils.escapeTagName(tagUnescaped);

	log.message(`Subscribing to tag ${tagUnescaped}`);

	try {
		// get the subscription
		const con = await connection;
		// subscribe to an item
		const monitoredItem = opcua.ClientMonitoredItem.create(con.subscription,
			{
				nodeId: opcua.resolveNodeId(config.nodeIdPrefix + tag),
				attributeId: opcua.AttributeIds.Value,
			},
			config.monitoredItem);

		// report new value
		monitoredItem.on('changed', dataArrived(tag, callback));
	}
	catch (err) {
		log.error(`Cannot subscribe to tag ${tagUnescaped}`, err);
	}
};
```
The following code throws an error on the monitoredItem.on('changed') line. It is taken from the 2.0.0 breaking changes wiki:
```
const addMonitoredItem = async (tagUnescaped, callback) => {
	// escape the tag name
	const tag = utils.escapeTagName(tagUnescaped);

	log.message(`Subscribing to tag ${tagUnescaped}`);

		// get the subscription
		const con = await connection;
		// subscribe to an item
		const monitoredItem = await con.subscription.monitor(
			{
				nodeId: opcua.resolveNodeId(config.nodeIdPrefix + tag),
				attributeId: opcua.AttributeIds.Value,
			},
			config.monitoredItem);

		// report new value
		monitoredItem.on('changed', dataArrived(tag, callback));

};
```
Here's an excerpt from the stack for the error:
```
at ClientSubscriptionImpl.monitor (/node_modules/node-opcua-client/dist/private/client_subscription_impl.js:147:29)\n
at eval (eval at thenify.withCallback (/node_modules/thenify/index.js:33:10), <anonymous>:12:12)\n
at new Promise (<anonymous>)\n
at ClientSubscriptionImpl.monitor (eval at thenify.withCallback (/node_modules/thenify/index.js:33:10), <anonymous>:10:8)\n
at addMonitoredItem (/monitored.js:30:48)\n    at process._tickCallback (internal/process/next_tick.js:68:7
```

erossignon, 2019-09-16:
OK,
Please replace
        
     const item1 = await subscription.monitor(itemToMonitor1, parameters1);

with 

     const item1 = await subscription.monitor(itemToMonitor1, parameters1, TimestampsToReturn.Both);

I'll fix the wiki

netbymatt, 2019-09-17:
That resolved it. Thank you.

closed 2019-09-17 by netbymatt