node-opcua/node-opcua #671: Connection not automatically reconnected

jgspiro, 2019-09-13:
**I'm submitting a ...** 

 - (x ) bug report => search github for a similar issue or PR before submitting

**Current behavior**
When reading or writing items, and connection is lost (server reboot), the connection is not re-established, and fails with "Invalid Channel" error at : 

```
at ClientSessionImpl.performMessageTransaction (/home/node/app/node_modules/node-opcua-client/dist/private/client_session_impl.js:669:29)
mqtt2opc_1     |     at ClientSessionImpl.performMessageTransaction (eval at thenify.withCallback (/home/node/app/node_modules/thenify/index.js:33:10), <anonymous>:6:48)
mqtt2opc_1     |     at ClientSessionImpl.write (/home/node/app/node_modules/node-opcua-client/dist/private/client_session_impl.js:375:14)
mqtt2opc_1     |     at ClientSessionImpl.write (eval at thenify.withCallback (/home/node/app/node_modules/thenify/index.js:33:10), <anonymous>:6:48)
```

We have another (OPCUA NodeJS) client that is connected.  Creating a new connection with yet another python client succeeds.

**Expected behavior**
Expecting the connection to be re-established when the channel is invalid.  Debugging the code it seems the invalid channel is not properly handled.

**Please tell us about your environment:**
OPCUA server : CodeSys
OPCUA Client : OPCUA NodeJS client version 2.1.5

  * (x ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: 
  * OS version: 
    - ( x ) Other   : specify : node:10-stretch docker image
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: CodeSys OPC server
     - Manufacturer/Software vendor:CodeSys
     
    
* **node-opcua version:**  : 2.1.5

* **Node:** 
   `node --version` = 10.16.3

erossignon, 2019-09-13:
Please provide client code for investigation

jgspiro, 2019-09-19:
Can't just post the entire source code.

closed 2019-09-19 by jgspiro

erossignon, 2019-09-19:
suggestion is to contact@sterfive.com for dedicated support in this case.